easytarget/esp32-cam-webserver #1: Headers too long error when using Apache proxy.

easytarget, 2019-11-20:
Also happens when I use a simple SSH tunnel, but NOT when i connect from my local network
https://github.com/espressif/arduino-esp32/issues/2983
I'd like remote access working...

GNDeSouza, 2019-11-25:
I have a similar problem.  Accessing from within my network goes well, but when I try from outside using port forwarding...  "headers too long..."

jorgechurriana, 2020-02-09:
Idem with Nginx 'proxy'

easytarget, 2020-02-10:
This is irritating; I spent some time trying to resolve this both via apache configuration, and then by (temporarily) editing the esp32 toolchain to allocate 4x as much space for incoming headers. 
Neither of these had any effect, unfortunately. 

Lack of time, and lack of need (this is nice to have, not essential, for me) mean I have not gone any further. Maybe my priority will change in the future, but for the present I wont do any more work on this unless I am very confident it will succeed.

easytarget, 2020-02-10:
Also note this comment: 
https://github.com/espressif/arduino-esp32/issues/2983#issuecomment-562304008

jumblies, 2020-07-19:
Same situation with traefik and file provider (to external server).  Oddly, this works with postman and similar apps and it frequently loads when using http rather than https

TungstenE2, 2020-09-29:
also got an error using my tablet with LineageOS 15.1 browser.
'Header fields are too long for server to interpret'

Using Chrome browser on the same tablet worked.

easytarget, 2022-03-07:
There may be hope for the next esp-ide-core release:
https://github.com/espressif/esp32-arduino-lib-builder/issues/52
https://github.com/espressif/arduino-esp32/issues/5969

00svd00, 2022-03-18:
For me on nginx this variant works

location /camera {
     proxy_pass         http://1.2.3.4:80;
     rewrite  ^/camera /(.*) /$1  break;
     proxy_set_header   Cookie "";
}
location /status {
     proxy_pass         http://1.2.3.4:80;
     proxy_set_header   Cookie "";
}
But no controls are working