easytarget/esp32-cam-webserver #2: Feature Request: Make vertical flip a rotate function.

Budman1758, 2020-01-03:
Just wondering if its possible to change the "V-Flip" button a rotate function in 90 degree increments.  Methinks that would be much more useful.  I have looked thru the code and can't find anything I understand to change it. My code skills are pretty much non-existent anyway.

This would make for many more mounting options for the camera.

easytarget, 2020-01-03:
Unfortunately this is not possible. You can flip the signal on both axes for a 180 degree rotation, but you cant rotate the image by 90 degrees, it is a hardware limitation, and here is nothing that can be done about it I'm afraid. You can mount the module upside-down, but not on it's side.

See [this too.](https://github.com/esphome/feature-requests/issues/174)

At a technical level, the camera module produces the video images directly (as a jpeg stream) and the esp32 module simply passes this already-encoded stream to the client. Any attempt to rotate the image would need to be done in software, and the esp32 lacks both the cpu speed, and memory, to do this in real time.

closed 2020-01-03 by easytarget

easytarget, 2020-01-03:
PS; good question :-)
Sorry for the quick reply and close but I'd looked at this myself too when coding this, and was a bit bummed that my mounting options would be limited too.

alejodeiuliis, 2020-08-28:
It's not possible to modify the html?

inside the index_ov2640_html_gz[]:

            .image-container {
                position: relative;
                min-width: 160px;
                -webkit-transform: rotate(90deg);
                -moz-transform: rotate(90deg);
                -o-transform: rotate(90deg);
                -ms-transform: rotate(90deg);
                transform: rotate(90deg);
            }

I coudln't make it work... don't know exactly why...

easytarget, 2020-08-28:
Ummm.. I guess I shpould have updated this ticket.
The code you want was introduced a few months ago in PR's #5 and #9 ; just pick up the latest version and set the rotation in the UI or even predefine it in `myconfig.h`