easytarget/esp32-cam-webserver #3: Proposal: script converting and compressing HTML pages automatically

karlitos, 2020-01-23:
Hi, thank you very much for providing this example code. Being actually a web-developer I was struggling with finding a way to be able to edit the HTML/styling more conveniently with some automatic conversion into the byte-arrays as in the original example.

I found some inspiration [here](https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/Camera/CameraWebServer) and after some tweaking I made a script utilizing the [bin2c](https://sourceforge.net/projects/bin2c/) tool to produce almost identical output as in the original example.

```bash
#!/bin/bash

OUTPUT_FILE=camera_index.h

convert_to_byte_array() {
    echo "Converting $@ to byte arrays"
    ./bin2c -m -o $OUTPUT_FILE $@
}

for file in `ls *.html`; do
    echo "Compressing: $file"
    cp "$file" "copy_$file" && \
    gzip -f "$file" && \
    mv "copy_$file" "$file"
    compressed_files=( "${compressed_files[@]}" $file".gz")
done
if test -f "./bin2c"; then
    convert_to_byte_array ${compressed_files[@]}
elif test -f "./bin2c.c"; then
    echo "Compiling bin2c from the source"
    gcc -o bin2c bin2c.c
    convert_to_byte_array ${compressed_files[@]}
else
    echo "Conversion of ${compressed_files[@]} to byte arrays not possilbe, missing bin2c.
Download bin2c source (bin2c.c) fom https://sourceforge.net/projects/bin2c/ and put it in this folder."
fi
````
I would like to replace the `bash` altogether with plain C code, but I wasn't able yet to figure out how to do the compression (some example is [here)](https://stackoverflow.com/questions/49622938/gzip-compression-using-zlib-into-buffer)

So, maybe you could be interested to use my bash script or help me with extending  the __bin2c__ to something like __html2c__ (There was [something](http://freshmeat.sourceforge.net/projects/html2c) 20 years ago 😄 )

easytarget, 2020-02-10:
Karel; I just realised I simply closed this without providing any explanation. Apologies for that, it was rude of me.
I basically think that developing such a script is a good fun exercise in scripting etc
.
But is it also re-inventing the wheel.. there are built-in tools in Linux that can do this already, I could generate a compressed and encoded header file very simply like this:
`gzip -kf index.html && xxd -i index.html.gz > index.html.h`
Adding that to a Makefile in a traditional development environment is easy, adding it to the Arduino IDE is hard.

Since this project has lots of free memory (note that the HTML is in PROGMEM, not RAM) it's not an issue here, and since the page is only loaded once the data saved by compression is insignificant vs the bandwidth needed for a video stream.

karlitos, 2020-02-11:
Hello Owen, thank you for your detailed response and I can fully understand your reasoning. Your explanation actually provides the EXACT solution for my original problem - I spend some time searching for what `xxd -i -u` is providing _out-of-the-box_ and I never found a single mention of `xxd` along the way 😃 

I think this would be a valuable addition to my [original stackoverflow post](https://stackoverflow.com/questions/59866735/converting-html-page-into-array-of-hexadecimal-values-for-use-in-an-arduino-serv) and I of course I will be mentioning you as the original author of this solution.

closed 2020-02-11 by karlitos

easytarget, 2020-02-11:
I cant remember where I heard of xxd either, just one of these random things I learned somewhere..

The reason why the simpler command you show produces different output is because when you run `xxd -i` without an input file (stdin) it *only* generates the hex content, and not the whole structure. When `xxd` has a input filename I uses that to generate full definition instead of just the contents.

See the effect for yourself like this:
```
$ echo "hello" > hello.txt
$ cat hello.txt | xxd -i
  0x68, 0x65, 0x6c, 0x6c, 0x6f, 0x0a
$ xxd -i hello.txt 
unsigned char hello_txt[] = {
  0x68, 0x65, 0x6c, 0x6c, 0x6f, 0x0a
};
unsigned int hello_txt_len = 6;
```

RobotHacker, 2020-07-04:
I had run across another option for encoding the html through Cyberchef [here](https://robotzero.one/esp32-cam-custom-html/).  From that I made the following recipes:
[decode](https://gchq.github.io/CyberChef/#recipe=Remove_whitespace(true,true,true,true,true,false)From_Hex('0x%20with%20comma')Gunzip())

[encode](https://gchq.github.io/CyberChef/#recipe=Gzip('Dynamic%20Huffman%20Coding','index.html.gz','',false)To_Hex('0x%20with%20comma',16))

easytarget, 2020-08-28:
Nice, A good solution for windows users I guess.

If I was making a commercial device with maybe dozens of complex html pages, javascript, graphics etc this would make sense. .. but If I were doing that I would also be using something more complex as my development environment, and I'd simply script something up with bash/xxd and put it as a pre-run script in my buildfiles. 

For this project the advantages of keeping the html trivial to edit far, far outweigh any advantages zipping and hex encoding it.