easytarget/esp32-cam-webserver #4: When discard face recognition features to allow OTA

jorgechurriana, 2020-02-09:
Would be posible to do a #define FACE in order to allow the option to compile without face recognition featrures in order to get a smaller code to fit in a OTA compiler option?

easytarget, 2020-02-10:
Good suggestion!
I'd be happy to recieve a PR for this, but it may be some time before I have the space to look at this myself.

easytarget, 2020-09-24:
Background:
Implementing OTA is quite easy, just add the libs and setup.
BUT; you need sufficient spare program memory to support it (eg uploading the new code alongside the old); and the face recognition makes this sketch too large.

jorgechurriana, 2020-11-28:
I have already done, but now i dont know how o mix with your code. I have other improvements, like flash and led control on the html, and eeprom wifi configuration saved

mark-hahn, 2021-05-29:
Do you know removing face recognition will shrink the bin enough?  When I installed the standard demo it used 83% of my space.  I created a partition table with no spiffs and it went down to 63%, still not small enough for ota.

easytarget, 2021-05-31:
It certainly should, it's a huge amount of extra code, and a number of other ESP CAm projects out there have enabled OTA after remove FR.

mark-hahn, 2021-05-31:
I was wrong.  I just installed an app (see below) that only serves a stream
and nothing else.  My 63% usage went to 19% so I have tons of room for ota.
 (This is still with no spiffs).

https://RandomNerdTutorials.com/esp32-cam-video-streaming-web-server-camera-home-assistant/

>
> It certainly should, it's a huge amount of extra code, and a number of
other ESP CAm projects out there have enabled OTA after remove FR.

easytarget, 2021-05-31:
I just ripped the face recognition out and added OTA. Tested and everything it took me <2 hours. LOL. It's now just 34% of storage used, and compiles much faster. Just pushed to on a branch called 'NoFace', I'm going to edit the readme since you ned to select a generic DEV board rather than the AI-THINKER one for this to work (AI-THINKER board uses a no-ota partition scheme).

mark-hahn, 2021-05-31:
> AI-THINKER board uses a no-ota partition scheme

Just specify another partition table in platform.ini.  It is also easy to make a custom partition file and drop it in your project folder.  I made one to get rid of spiffs.  I'm going to do another one that is ota with no spiffs.

easytarget, 2021-06-01:
> Just specify another partition table in platform.ini

Yes, it is easy enough to do this for platformIO, but this project is for the Official Arduino IDE and so the instructions need to be suitable for that.

mark-hahn, 2021-06-01:
My bad.  I should point out that even for platformio it isn't as easy as dropping it into the project folder.  There are some other magic incantations needed.  I just found this out the hard way.

jorgechurriana, 2021-06-02:
Hello, yes i have removed face recognition and enable standard OTA and is working

easytarget, 2021-06-08:
I just dropped the 4.x beta with working OTA and no face recog. .travis and Platformio.ini are updated as appropriate, as is the README.

closed 2021-06-08 by easytarget