easytarget/esp32-cam-webserver #7: Any way to have settings saved upon reboot in Preferences

Alexplose, 2020-04-08:
I'd like to have SSID & key added to settings panel, with a fallback and also all of theses settings saved in Preferences, to restore them on reboot.

Do you plan to do that ?

easytarget, 2020-04-10:
As described; No. But I am going to create two new issues since you are on the right track ;-)

When I think about this I can see two separate requests here ; 1) saving camera settings and 2) adding Wifi management functions. 

For saving settings the issue is fairly straightforward.

For WiFi connecting, reconnecting, saving and falling back to AccessPoint mode I would like to use one of the frameworks that already exist for this; There is no sense in reinventing the wheel here.
- And.. I really want to combine adding WiFi manager features with enabling OTA firmware updates; having to remove and hook modules up to my programmer is a total pain.

Alexplose, 2020-04-15:
Thanks for your answer, and yes there is two separate requests, saving the settings is the easiest part here.
For the second one it requires a little more writing.

easytarget, 2020-09-09:
Doing some cleaning; this covered by #20 so closing

closed 2020-09-09 by easytarget