easytarget/esp32-cam-webserver #21: face recognition at startup

trevor-sonic, 2020-08-29:
Hi there,
I liked your improvements. I'm actually looking for a startup configuration with face recognition. I mean when cam powered on  automatically face recognition must be selected. Would you add this in your improvement list?

easytarget, 2020-08-30:
Yes, that is exactly the sort of thing I'd like to put into defaults. I'll see what I can do. I have an issue tracking this, #18 
Note, that since Face recognition requires low resolution it makes sense to combine with setting the default image resolution.

easytarget, 2020-09-09:
Fixed by #24 ; you can enable detection and recognition by default in myconfig.h.
unfortunately face enrolments are held in ram, and lost between reboots. It will take work to save them to LittleFS/SPIFS, see #20

closed 2020-09-09 by easytarget