easytarget/esp32-cam-webserver #28: Face recognition Database

easytarget, 2020-09-09:
Can this be saved/restored?
* Need to check; it appears to just be a indexed array/structure in memory, so I assume the answer is yes.

If so:
* Save/Restore to LittleFS
* Upload/Download from the clients browser

trevor-sonic, 2020-09-11:
Default face recognition functionality is very basic and useless without ability  to reuse  after rebooting.
Ability to Save/Restore would be fundamentally needed.

easytarget, 2020-09-11:
I agree, but I also consider face recognition a low-priority in this project. I fully intend to remove it completely in V5 so that I can put OTA updates in.
I've just expanded this request, and added the 'Help wanted' label, I'd love to work on it but I need to put this project aside for a while after v3..

easytarget, 2020-09-11:
The LittleFS part of this is common to #20

trevor-sonic, 2020-09-11:
OTA is important from programmer point of view but face recognition is very unique feature compare to other ip camera on the market (I bought it just for this). I would say please don't remove (of course this is your repo, decision ), or keep a git branch with it.

easytarget, 2020-09-11:
Dont worry, there will be a v4 dev and maintenance branch..

easytarget, 2020-09-25:
`spiffs` branch is a POC for preferences saving; but the initial commit there also shows how to read/write a C `struct` to spiffs. This should be doable for the face recognition db; sabing `mtmn_config` and `id_list`. (declared as static [here](https://github.com/easytarget/esp32-cam-webserver/blob/master/app_httpd.cpp#L79))

easytarget, 2020-09-27:
I just merged #49 which, as well as the preferences save/restore includes stubbed-out code and commented out html for the FaceDB operations and UI.

It also has Debug dumping of some facedb data sizes, I lost several hours going through the esp-face code without really getting a handle on how this works. It looks like the structures are dynamic with static indexes. 

Needs more thought.

easytarget, 2020-09-28:
https://github.com/espressif/esp-face/blob/master/face_recognition/fr_flash.c
Saves/reads from flash memory; *but* needs a change to the board partition scheme and I wish to keep this project simple enough to use with a unmodified Arduino IDE.

otoh; it shows me what I need to save; and how to save it to a block device on a partition.. Good basis for a SPIFFS implementation.