easytarget/esp32-cam-webserver #37: Save images to SD card

easytarget, 2020-09-11:
V4 extended..
Allow images to be (timestamped and) saved to a folder on an attached SD card (AI-THINKER has a SD card slot)
* Trigger button in UI
* Timelapse?
* Face recognition triggers?
  * Named faces
  * Unknown faces
Ideally, allow images to be browsed and downloaded via the WebUI.
* as a bundle? might be easier for UI..

bill-orange, 2020-10-04:
This is an enhancement I am also looking for.  I may work on the code for this myself but I am likely to make it too much of a kluge.

easytarget, 2020-10-04:
Oops.. fat fingers. Didnt intend to close there. 
Thiis would be nice, but very very low priority for me right now.
There are a number of other projects that can do saving, loading, downloading and even movies on the SD card. 
In particular check out:
https://github.com/jameszah/ESP32-CAM-Video-Recorder

sjude68, 2021-01-13:
I clicked the "Enroll" button but nothing happens.  Does it save the captured image on the SDCard.  How do I see the images on the SDCard and add new images for Face Recognition.

easytarget, 2021-01-21:
@sjude68 Unfortunately, you cant. That's why this issue exists and is not resolved. 
The enroll button simply enrolls faces in the face recognition db, which is held in memory, lost on reboot, and contains face data not images.

easytarget, 2021-07-06:
(from #135 ) If implementing check out the comments in: https://github.com/espressif/esp32-camera/issues/203

abratchik, 2022-12-03:
Fixed in PR #280. Please check this [fork](https://github.com/abratchik/esp32-cam-webserver) - all the web pages ported to SD