easytarget/esp32-cam-webserver #52: http requests

TungstenE2, 2020-09-28:
Hi all,

I understand that in the original demo version some http requests are supported to set parameters.

Is setting the light by http request supported in this version?

I would like to do the following from my FHEM home automation.

- if door bell pressed
- http request to turn on LED (if night)
- get still picture
- send still picture via Telegram
- http request to turn off LED

thx for your support

easytarget, 2020-09-28:
Yes; **all** browser/cam communications are via http request; So you can trigger absolutely everything with properly crafted HTTP requests.  

I'm not near my desk now, so this is a guess from memory:

Snapshot is something like `http://IP/snapshot'  and settings are something like `http://IP/cmd?val=lamp,ret=100  (where the last value is in percentage, 0=off, 100=full). I've probably got the exact syntax wrong.. I'll double-check later.

The snapshot URI is in a tooltip when you hover on the snapshot button, and the settings commands that are sent are displayed in the browser developer debug console, so if in a hurry open the developer tools and check out the console, and the URL's it requests.

TungstenE2, 2020-09-28:
thx, snapshot is already working... ;-)

TungstenE2, 2020-09-28:
found it!
http://IP/control?var=lamp&val=100

thx

easytarget, 2020-09-28:
Cool; I just spawned three new issues as a result of thinking about this.. when will I ever learn. ;-) 
anyway.. closing.

closed 2020-09-28 by easytarget