easytarget/esp32-cam-webserver #56: feature request: more system info

TungstenE2, 2020-09-28:
Please implement some additional system info:

Could think of the following:

- Local Time
- Uptime
- Load
- Free RAM
- Free Stack
- RSSI

[image]

easytarget, 2020-09-28:
Good idea; simple enough to do on a dedicated URI I imagine; I would also add:

Module Temperature (esp32 has internal temp, useful for monitoring the CPU)
VCC ([readVcc](https://www.letscontrolit.com/forum/viewtopic.php?p=2597&sid=5c0464f03adb95d183a83409699246b9#p2597))
Internal [Hall sensor ](https://www.esp32.com/viewtopic.php?t=4963#p32861)
And any free GPIO lines? Trickier; maybe just show how it can be extended in code.

Also being able to return this data in a JSON string (add ?json=true to URI?) so that it can be parsed by home automation systems would be a nice to have

easytarget, 2020-10-03:
Ok; s status page is implemented on `/dump`; linked from the 'Name' field in the UI.
- No Temp or VCC, unfortunately they are depreciated on later ESP32 models and give error results (yay! thanks ESP, and thanks doubly for documenting that so very well.) /S)
- The hall sensor fluctuates a lot and is not very sensitive; decided not to show this.
- Uptime is shown, but not the RTC, that can be added if/when NTP is implemented ( #38 )
- getting a CPU 'load' is really needs RTOS or similar so you can count 'idle' time as a separate task.

commit adfa3ca8 (2020-10-03): Stream view fix, Status dump (#63)

* Better treatment of rotated stream in stream viewer,
still not perfect but ok for now.
* Info Page: fix #56
closed 2020-10-03 by easytarget