easytarget/esp32-cam-webserver #57: video stream using tinyCam PRO is not working

TungstenE2, 2020-09-28:
Hi, 

when I used the original demo version available in Arduino IDE I was able to get the video stream on my mobile using tinyCam PRO Android app. 

Settings:
Vendor: Generic
Camera model: Generic URL
URL: http://IP:81/stream
Protocol: snapshot (JPEG)

Now with your project the above settings (but URL http://IP:81/view) does not show the stream.

Also changing the protocol to 'Server push (MJPEG)' does not work.

[image]
[image]

Any ideas?

Thx

easytarget, 2020-09-28:
Remove the `view` part of the url, eg `http//IP:81/`
the stream itself is on the raw port, the `/view` URL is for the html page of the stream viewer

TungstenE2, 2020-09-29:
great, this worked!

May be it would worth to have a URL chapter on the project info page, listing the most common once:

eg:
GUI for settings
http://"IP"/

Snapshot
http://"IP"/capture

Stream in html page
http://"IP":81/view

raw stream for survillance clients
http://"IP":81/

commands:
Flash light: on/off
http://"IP"/control?var=lamp&val=100
http://"IP"/control?var=lamp&val=0

etc...

easytarget, 2020-09-29:
That is why I created #55 :wink:

closed 2020-09-30 by easytarget