easytarget/esp32-cam-webserver #58: feature request: implement fallback SSID

TungstenE2, 2020-09-29:
Hi all,

how about implementation off a fallback/alternative SSID for WIFI?

ESPeasy and Tasmota are supporting this for ESP8266, so this should be possible with ESP32 s well, right?

Would be a great settings option.

[image]

easytarget, 2020-09-29:
A sensibe idea, but this is not in my immediate 'hit list', as usual I'm happy to receive PR's if somebody wants to give it a shot.

To be honest I'd rather do something akin to implementing a proper WiFi framework (there are several floating around) that allows for learning multiple station's and their settings in the same way as your phone or laptop, and falls back to AP mode for initial setup if it cannot connect. It also makes sense to combine doing that with OTA ( #4 ) and Uploadable HTML files ( #29 ).
- My [GRBL-ESP32](https://github.com/bdring/Grbl_Esp32) based CNC controller works that way, and it's lovely...

TungstenE2, 2020-09-30:
to be honest I disagree somehow. Having the device going into AP mode in cases wifi is disconnected caused me some trouble on wifi switches. Tasmota has options to control this. eg Tasmota option 5 works best for me. Setting of 2 wifis is possible.

Tasmota wifi options:
WifiConfig		Show current config tool
WifiConfig	0	Disable wifi config but restart (used with alternate AP)
WifiConfig	1	Start smart config for 1 minute and set as current config tool
WifiConfig	2	Start wifi manager (web server at 192.168.4.1) and set as current config tool
WifiConfig	3	Start WPS config for 1 minute and set as current config tool
WifiConfig	4	Disable wifi config but retry other AP without restart
WifiConfig	5	Disable wifi config but retry same AP without restart and flash writes

easytarget, 2020-09-30:
Everybody has their preferences ;-)
Tasmota is exactly the sort of framework I was thinking of. But It's not on my immediate roadmap.
Something involving WifiMulti() and being ablt to supply multiple SSID/pass combos might be much easier to implement. But it's still on the back burner. sorry.

easytarget, 2020-10-06:
Ok; Please see #64 ; you can define multiple ssid/passwords and the strongest will be selected. Re-scans on disconnect. AP can be the default, or a fallback, or disabled entirely.
:grinning:

closed 2020-10-06 by easytarget