easytarget/esp32-cam-webserver #59: stream freezes

TungstenE2, 2020-09-29:
Hi all,

stream freezes regulary.
[image]

Powered via USB power bank 5,17 V.

Please check

easytarget, 2020-09-30:
I'm afraid mid-frame lockups are pretty inevitable unless your network is perfect. When developing I see them all the time, but no associated errors in serial debugger or browser console to indicate anything failed in the code,

With the chunked stream used here an active connection has to be held from the browser to the cam, and the cam then blindly pushes data over that. If the connection drops for any reason the stream freezes. 

Changing the stream protocol could help but that is a whole other topic. Other ESP cam solutions exist if this is critical. For the present I'll keep the simple streaming protocol as implemented bu expressif. Another option could be code in the browser that detects a failed stream and re-starts it.

easytarget, 2020-10-04:
https://github.com/espressif/esp32-camera/issues/162 may be related.

on the other hand, I've had a webcam streaming to to a pi running ffox; and running continually (stream still working) for 2+ days now in my workshop. I'll try increasing the resolution and light levels to see what that does.