easytarget/esp32-cam-webserver #60: Small fix in storage.cpp

benzino77, 2020-09-30:
Hi there!
Thanks for a great project focusing on important things for many of us.

I will not make a pull request for that, but I do think that in file storage.cpp in line 2
https://github.com/easytarget/esp32-cam-webserver/blob/master/storage.cpp#L2

it should be:
```
#include "esp_camera.h"
#include "jsonlib/jsonlib.h"
#include "storage.h"
```
instead of:
```
#include "esp_camera.h"
#include "jsonlib/jsonlib.cpp"
#include "storage.h"
```
(I'm using platformio on VSCode and it throws error when the linker is started at the end of compilation)

easytarget, 2020-09-30:
I just 'addressed' this in a commit; but I think it is still broken for platformio; but that should be solvable in the settings file (specify include dir).

I'm just installing platformio myself, I'll try and get this building with that and see what I end up with. 
- Ideally I want a PlatformIO build to happen as part of the travis build, to can catch breakage.

easytarget, 2020-09-30:
This seems to work now; same codebase builds on both PlatformIO and ArduinoIDE

closed 2020-09-30 by easytarget

benzino77, 2020-10-01:
More platformio style directory structure should looks similar to this:
[image]

Where
include directory is intended for project header files.
lib directory is intended for project specific (private) libraries.

easytarget, 2020-10-01:
Yes; and if doing a platformio project from scratch I'd `pio project init` and accept the default structure. It's a good structure for an ongoing project as it grows.

But this is a Arduino IDE project aimed at tinkerers, not software engineers, so I compromised. Opening the repo root in the Arduino IDE opens all the important project files as tabs in the conventional way, and, as a bonus, PlatformIO also works out of the box. I might even get round to supporting the Expressif IDE at some point.. but the layout wont be changed for that either.