easytarget/esp32-cam-webserver #61: Components-Object deprecated and to be removed

TungstenE2, 2020-09-30:
I am getting a warning in Firefox deveopper tools about deprecated Components-Object still used.
It should no longer be used.

'Das Components-Objekt sollte nicht mehr verwendet werden. Es wird bald entfernt.'

[image]

easytarget, 2020-09-30:
Humm, curious.
Which FFox version are you using; and what OS (Win/Linux?) 

I dont see any depreciation warnings, nor any of the CSS warnings. I'm on Fedora32 and update daily. I just double-checked and have a clean log on FFox 81.0, Chromium 85.0.4183.102, Chrome 86.0.4240.53 and Chrome(unstable) 87.0.4270.0

TungstenE2, 2020-10-01:
using FF 81.0 64-Bit on Windows 10

currently using 2 cams, same build date, but differnt lenses.
Both have the warnings.

[image]

easytarget, 2020-10-01:
Thankyou, I'll reboot into windows later and see if I can reproduce,

easytarget, 2020-10-03:
I do not see this (Win10, ffox 81), nor in edge etc.. Looking at the error I noticed it is *not* referencing the html or css, but is attached to the IP address itself

On my system I see:
[image]
I get the other warnings about pseudo-classes etc (they will be fixed), but the big scary depreciation notice is absent.

I'm guessing this is something specific to your browser extensions, I found this:
https://stackoverflow.com/questions/59107409/what-to-do-about-the-components-object-is-deprecated-it-will-soon-be-removed

Closing for now.

closed 2020-10-03 by easytarget