easytarget/esp32-cam-webserver #66: V3 prerelease2: boot error

ozett, 2020-10-06:
i had this before on the usb-port of my PC, 
not enough current...

but with same module and pre-release2 (beta3) version,
it does not recover. even on a stronger USB-port.

something i can do about?

```
===
esp32-cam-webserver: ESP32 camera server
Code Built: Oct  6 2020 @ 14:25:17
Camera init succeeded
OV2640 camera module detected
Internal filesystem contents
Listing SPIFFS directory: /
Preference file /esp32cam-preferences.json not found; using system defaults.
Lamp: 0%, pwm = 0
Starting WiFi
Known external SSIDs: None
MAC address: CC:50:E3:95:12:24
Scanning local Wifi Networks

Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

====
esp32-cam-webserver: ESP32 camera server
Code Built: Oct  6 2020 @ 14:25:17
[E][camera.c:205] skip_frame(): Timeout waiting for VSYNC
[E][camera.c:1270] esp_camera_init(): Camera init failed with error 0x20003
Halted: Camera sensor failed to initialise
Will reboot to try again in 10s

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

====
esp32-cam-webserver: ESP32 camera server
Code Built: Oct  6 2020 @ 14:25:17
[E][camera.c:205] skip_frame(): Timeout waiting for VSYNC
[E][camera.c:1270] esp_camera_init(): Camera init failed with error 0x20003
Halted: Camera sensor failed to initialise
Will reboot to try again in 10s

```

easytarget, 2020-10-06:
```
Scanning local Wifi Networks

Brownout detector was triggered

```
That's your problem, right there. 
When the WiFi scan starts themodule is drawing more power and the voltage dips too low, aka a 'brownout', it is a hatrdware failure that cannot be fixed in code.

Doing a network scan is a high power operation, so it's entirely possible that the code worked before, but starts to fail now, and the following 'camera sensor failure' is a follow-on from that, it has browned out too.

Doublecheck your wiring.. thin long wires can cause the voltage to drop as the unit draws more power, as can weak power supplies and dodgy solder/crimp connections. If it is any comfort, you are not alone. I had to beef up the wiring for one of my cams.

Another thing I have seen others doing is to add a big(ish) electrolytic capacitor across the 5v line to smooth out spikes/dips. This seems to help with interference on the video too by keeping the camera sensors voltage constant. Use a 220/470uF electrolytic and get the polarity right ;-)

ozett, 2020-10-06:
thx for the infomation. i guess is pushing me in the right direction.

i had problems with the left one as shown on picture
(on PC-usb port and on some powerfull usb-hub too - i will change this now again),
and i will also try if the other module works better on only 2 wires 5V...

after that, i will try a complete other powersource -- 
and not questioning the firmware for sometime  😉  , or until release 👍  😉  ...
thx

[image]

easytarget, 2020-10-06:
I think that this is because the WiFi scan uses a lot more power than the previous 'start and just listen' mode..
It's quite common, I found a lot of similar issues, eg: https://github.com/espressif/arduino-esp32/issues/863

ozett, 2020-10-06:
i flashed it again on the usb-port on my PC (where if failed before),
but got luck this time... ( i will keep tryiing..)

[image]

ozett, 2020-10-06:
third try and success.
now the same firmware and modul is running on usb-port on same pc as before,
and also on powersupply..

[image]
[image]

easytarget, 2020-10-06:
Oops. Fat fingers.
Glad you got it working.

easytarget, 2020-10-06:
I'm guessing you only had one entry and still had `WIFI_AP_ENABLE` on, 
The output in console is not very clear when configured like that' I'll make a small fix so that scan is suppressed and a proper 'AP Mode On' message emitted.
I've also worked on the commenting and layout of the list definition. This is still a work in progress so apologies for all the confusion.

closed 2020-10-06 by easytarget