easytarget/esp32-cam-webserver #67: V3 ++ Versioning?

ozett, 2020-10-06:
i try V3beta3, and before the github version.
How do i differentiate the code-base/versions?
maybe even if it is flashed in different versions on different boards?

when the board dont get a webgui ready,
 it would be great to see the version on the serial-monitor..

ozett, 2020-10-06:
i think i found it 
(on the old github version, because no reboots like in v3beta3)

where can i check it in the source? i made several folders with versions, i cannot see on the folder which is which..

[image]

easytarget, 2020-10-06:
The best I can (automatically) provide is the build time that you found. And this is not recorded in the code, it is added when you compile it.

I'll think about adding a version to the code; but I'll have to remember to update this myself or set up some complicated scripting on my build machine(s). Yet one more thing to maintain/go wrong... :frowning_face: but you are correct in that it would be a useful thing to supply.
- Side step to rant about the ArduinoIDE/toolchan not including a git describe macro or similar...

Until then you'll need to keep track yourself of where you unpacked the releases. Sorry.

ozett, 2020-10-06:
i played before with espeasy a lot https://github.com/letscontrolit/ESPEasy

may you have a look into their code?
they have a statuspage with somethin like `git-versioning`.. 

dont know how they do it, maybe an idea to investigate?
(still also not fully implemented versioning at their side)

[image]

easytarget, 2020-10-06:
And it is also trivial in platformio, VS, esp ide and any commandline tool.. 
However.. 
This project targets beginners and the Arduino IDE. Which has no equivalent. So it's not going to 'just happen'.

easytarget, 2020-10-12:
Sorry if last reply was a bit terse; I've got a lot on.
But.. fundamentally a good suggestion, so I have added a version string (that I need to update manually) to the code and outputs.

PR #71 (2020-10-12): Simple versioning system
Add a simple upstream versioning system, string reported in startup console + dump output. Fix #67 

With the ArduinoIDE this is about as far as it can be automated.. but should assist in debugging user issues etc.
closed 2020-10-12 by easytarget