easytarget/esp32-cam-webserver #72: Suggestion: Add Event Driven Illumination LED Control

TodWulff, 2020-10-14:
Good day.  Thank you for your effort in developing this and putting it out there for the masses.

It is proving useful for my employment of the ESP32 CAM as a 3D printer camera and is beneficial in this use case as I can set default resolution and have the Illumination LED on.

It would be useful to be able to have the Illuminator LED off when not streaming but allow a preset illumination level whenever a capture event transpires vs. having it on and generating heat all the time.

Thanks again!  Take care.

easytarget, 2020-10-14:
Nice idea. It also has application as a stream on/off indicator when set to a very low value..

easytarget, 2020-10-25:
Implemented and controlled by a web setting, which is saved in preferences along with lamp level. :smile:

closed 2020-10-25 by easytarget