easytarget/esp32-cam-webserver #74: hostname

mrmessy, 2020-10-20:
I had to add a couple of lines of code to specify the hostname.
```cpp
WiFi.config(INADDR_NONE, INADDR_NONE, INADDR_NONE);
WiFi.setHostname("my_cam_hostname_goes_here");
```
I'm not sure where the best place to put them is but they work here, at the top and bottom of this snippet from the main file:
```cpp
    WiFi.config(INADDR_NONE, INADDR_NONE, INADDR_NONE);
    if (bestStation == -1) {
        if (!accesspoint) { 
            #if defined(WIFI_AP_ENABLE)
                Serial.println("No known networks found, entering AccessPoint fallback mode");
                accesspoint = true;
            #else
                Serial.println("No known networks found");
            #endif
        } else {
            Serial.println("AccessPoint mode selected in config");
        }
    } else {
        Serial.printf("Connecting to Wifi Network: %s\n", stationList[bestStation].ssid);
        if (stationList[bestStation].dhcp == false) {
            #if defined(ST_IP)
                Serial.println("Applying static IP settings");
                #if !defined (ST_GATEWAY)  || !defined (ST_NETMASK) 
                    #error "You must supply both Gateway and NetMask when specifying a static IP address"
                #endif
                IPAddress staticIP(ST_IP);
                IPAddress gateway(ST_GATEWAY);
                IPAddress subnet(ST_NETMASK);
                #if !defined(ST_DNS1)
                    WiFi.config(staticIP, gateway, subnet);
                #else
                    IPAddress dns1(ST_DNS1);
                #if !defined(ST_DNS2)
                    WiFi.config(staticIP, gateway, subnet, dns1);
                #else
                    IPAddress dns2(ST_DNS2);
                    WiFi.config(staticIP, gateway, subnet, dns1, dns2);
                #endif
                #endif
            #else
                Serial.println("Static IP settings requested but not defined in config, falling back to dhcp");
            #endif
        }
        // Initiate network connection request
        WiFi.setHostname("my_cam_hostname_goes_here");
        WiFi.begin(stationList[bestStation].ssid, stationList[bestStation].password);
```
The hostname probably needs a myconfig.h setting too.

I think I found the solution here https://github.com/espressif/arduino-esp32/issues/3438

easytarget, 2020-10-20:
Interesting; would be nice to add this (I'm using IP addresses at home.. but it would be nice to supply this by default too)
Adding a option to control whether the UI uses hostname or IP address for display + links should really be part of this too.

easytarget, 2020-10-25:
Implemented as option in 'myconfig'

closed 2020-10-25 by easytarget