easytarget/esp32-cam-webserver #76: v3.0-RC2: issue with switching APs

TungstenE2, 2020-10-28:
Hi,

I installed v3.0-RC2 on two of my cams, one with external antenna.

I have 2 AccessPoints with the same SSID. Usually my devices connect to the AP with the strongest signal
but the two cams are only connecting to the AP which they connected to at first connect.

Does the cam store information of the AP somehow?
I once had this with another IP cam, where the cam refused to connect to another AP with the same SSID.

thx

easytarget, 2020-10-28:
Interesting.. I assume it shows both AP's as 'known' in the serial console list. But connects to the weaker signal?

Currently the code just notes the SSID name of the strongest signal and then the actual connection is established with a `WiFi.begin(ssid,passwd);`. So if Wifi.begin() simply takes the 'first' match it finds (and dosnt look for signal strength etc) it may well prefer the wrong AP.
 [I believe it has some logic for 'fast connecting' that may be making it use internally and complicating things too.. but I'm not really sure.. :frowning: ]

Anyway... I think I can see a way to aleviate some of this pain by Connecting to BSSID instead. This isn't the same as doing a full ESS enterprise multi AP support thing (which is not well supported in the Arduino esp32 libs).

I'll look into the code later, but just as a proof maybe you could replace line 230 in `esp32-cam-webserver.ino` with the following two lines:
````
                String thisBSSID = WiFi.BSSIDstr(i);
                Serial.printf("%3i : [%s] %s (%i)", i + 1, thisBSSID.c_str(), thisSSID.c_str(), thisRSSI);
````
You should see both AP's listed in the 'Found Networks' serial output. With different BSSID's.
the esp32 implementation of WiFi.begin() *is* extended to allow specifying the bssid (hurrah :tada: ) but I need to think further on the logic for this, since I need to cater for AP mode too.
Also, I'd like to allow specifying the BSSID in place of SSID in the actual station list.. which could be handy.

TungstenE2, 2020-10-28:
hm, seems like today cam switched the AP by itself. This morning it started with the weak AP an now it is connected to the strong AP. strange...

easytarget, 2020-10-28:
I'm not that surprised, this is what I mean by the begin() not caring about relative strengths, it's probably just selecting on a randomish 'order of discovery'. The BSSID stuff i have in progress should make it default to the strongest bssid.

easytarget, 2020-10-29:
Hokay; I just committed a fix for this to the master branch, it will be in 3.0, it records both the SSID and BSSID when scanning for the strongest network; and then connects with both parameters supplied, if two SSID's have the same name it will now chose the strongest regardless of scan order.. It also allows you to specify a BSSID in place of the SSID in the array so you can force a specific AP if desired.

Also; I took the opportunity to split the definition for the station list, to make it a little less confusing.

closed 2020-10-29 by easytarget

TungstenE2, 2020-10-29:
great, thanks!