easytarget/esp32-cam-webserver #80: Password protected server

mungle, 2020-11-03:
I was trying to password protect the web stream following this link:
[https://microcontrollerslab.com/esp32-password-protected-web-server-in-arduino-ide/](https://microcontrollerslab.com/esp32-password-protected-web-server-in-arduino-ide/)
 but I'm not really able to find the piece of code where to implement this.

Any help, suggestions, alternatives? Thank you!

easytarget, 2020-11-08:
Hi, thanks for the question.
I struggled with the example there for a bit (but it works, I tried it out), but see it now. The secret is returning the `401 Unauthorized` followed by `WWW-Authenticate` in response to the first connection attempt (lines 76/77), which triggers the authentication prompt when needed.

Anyway.. the short answer is to strongly emphasise that this cam server has no secure wifi connection, so any serious snooper will be able to trivially sniff your connections and bypass any protections you may establish. Until SSL is implemented ( #54 ) the best you are doing is guarding against unsophisticated casual access.

The longer answer is that you can modify the stream and capture handlers in `app_httpd.cpp` with the same authentication test and response. When the handler is called you initially test the request headers and send the appropriate 401 response if they lack authentication. This would protect the images but leave the control pages unprotected.
To protect the whole server you would need to do this to /every/ handler for all the various pages. At which point it is better to look at some of the more complex solutions with a separate login page and cookies used to authenticate after login.

easytarget, 2020-11-19:
No followup, covered by #54 . closing.

closed 2020-11-19 by easytarget