easytarget/esp32-cam-webserver #81: Compile error

projectsun2, 2020-11-04:
What have I done wring?

[image]

easytarget, 2020-11-06:
Looks like you do not have the correct version of the ESP32 libraries.
I only test/support building against the latest ESP32 libraries from https://dl.espressif.com/dl/package_esp32_index.json 
- Currently that is v1.0.4
[image]

If you have a older version, upgrade..
If you got the ESP board via another method, or have modified the installed files in `.arduino15/packages/esp32/` you'll need to undo that (at least while you wish to compile this code..)

projectsun2, 2020-11-06:
Issue can be closed: Yes, I had 1.0.1. Updating did the trick. Thank you.

easytarget, 2020-11-07:
Glad you got it going. thanks for the feedback.

closed 2020-11-07 by easytarget