easytarget/esp32-cam-webserver #82: Serial monitor fuckup

NodeSolo88, 2020-11-07:
I have had many problems with my esp32 cam and have solved them but there is one I cant figure out. I have used the FTDI USB to TTL converter and two Arduino uno's and nothing is working. After running the camera web server script and removing the jumper from the esp32 cam I would open the serial monitor and click the RST button to get the ip adress of the cam. But instead of the ip adress this error would pop up: 

[E][sccb.c:154] SCCB_Write(): SCCB_Write Failed addr:0x70, reg:0x12, data:0x80, ret:-1
[E][sccb.c:154] SCCB_Write(): SCCB_Write Failed addr:0x70, reg:0xff, data:0x00, ret:-1
[E][sccb.c:154] SCCB_Write(): SCCB_Write Failed addr:0x70, reg:0x05, data:0x01, ret:-1
[E][camera.c:1215] camera_init(): Failed to set frame size
[E][camera.c:1270] esp_camera_init(): Camera init failed with error 0x20002

If anybody could help me that would be greatly appreciated.

easytarget, 2020-11-07:
Google can help, a bit. I2C comms are failing.
First thing to check is that you are powering the module from 5v, not 3.3v. Be careful.. dont simply switch the ftdi back to 5v, disconnect the data pins first.
Other than that it may be a bunch of stuff; bad board selection, poor components.. (does the original example script from the esp32 library work?) it could even be dodgy soldering on the board etc.. I see a lot of similar errors in a simple search, and a lot of different explanations.

easytarget, 2020-11-07:
nb. I'm not prudish but some search engines and others are; so keeping language mild. ;-)

NodeSolo88, 2020-11-07:
> Google can help, a bit. I2C comms are failing.
> First thing to check is that you are powering the module from 5v, not 3.3v. Be careful.. dont simply switch the ftdi back to 5v, disconnect the data [...]

Thank you so much for your help. The weird think is I was using the 5 volt connector and I couldn’t find anything on google for my issue. I ended up solving it buy putting a code in that disabled the brownout sensor and that fixed it. I think I had some hardware issues because it didn’t work with the usb converter that came in the kit and I was having so manny other problems. I even programming it with my arduino uno and wall power to make shure it was getting enough power but the brownout sensor was still triggered. Again Thank you so much for your reaching out

closed 2020-11-07 by NodeSolo88

easytarget, 2020-11-08:
Thank you too for letting me know.
There does seem to be alot of variability between boards and camera sensors, power issues/sensitivity are especially common.