easytarget/esp32-cam-webserver #85: no stream possible anymore

TungstenE2, 2021-01-04:
Hi,

I have 3 ESP32 cams, all operation with your webserver version.
Compiled Oct 26 2020 @ 12:42:11

One of my cams is acting strange for a couple of weeks.

Webserver is working.
Get Still is working but no streaming is possible.

Auto lamp is always switching on the LED, like as a stream is currently in progress, but I have checked all my devices and none is calling the stream.

Any idea how to fix this? Device is powered 24/7 in a 3D printed housing.

Thx

TungstenE2, 2021-01-04:
Seems I found the reason in my network and could fix it.

But on the way I discovered that the webserver will only connect to channel 1 and ignores closer network with stronger signal and same SSID but using channel 6.

closed 2021-05-09 by easytarget

buteomont, 2021-12-18:
> Seems I found the reason in my network and could fix it.

Care to elaborate? I'm having the same issue.  I can control the light, etc. and can take still snapshots but the streaming will not work.

easytarget, 2021-12-18:
The stream happens on a different port number (port 81 by default); so I'm guessing it was a NAT/route/firewall issue somewhere along the line ;-) or possibly something even more gnarly and complicated.

If you still see an issue, try directly connecting to port 81 (the URL is in the serial startup dump) and see if this works.. 
- If you cant connect to it then you probably have a network issue and need to address that.
- If you connect but get no response have a look at the serial debug info, (send a 'd' in the console for a dump too) and start a new issue if you cant find any good answers online or in this issue tracker.

buteomont, 2021-12-18:
Thanks.  I can get the stream on port 81 directly using VLC, Opera or 
Firefox, but the stream won't show up on the web page with the controls. 
Might be a CSS problem since I can never get mDNS to work on my network and 
I have to use the actual IP address in the URL. That's not a big deal but 
the frame rate is really slow, maybe one frame/second. Not sure if that is 
normal for this camera.