easytarget/esp32-cam-webserver #86: Add timestamp to capture

drtorchwood, 2021-01-09:
Hi there, 

I use my unix machine to take regular captures with `wget -O capture_$(date) <ip>/capture`. In principle, this is fine but I would be great if the time would not only be part of the filename but directly printed in the picture. 

Perhaps (if at some time an ntp module is included) this feature could be added. Thanks!

easytarget, 2021-01-21:
Good suggestion, I'd considered this before as I started this project, and decided it was not possible due to memory constraints.

But now I know the code better; I think it is entirely possible to write to the framebuffer before sending images and streams. This is (after all) done for face recognition to put boxes and labels on detected items. So the code is (sorta) already present.

Something for V4, I'll add this to the issues for that.

Meanwhile...

I also capture images on my home server, I do the timestamping on the server side; 

Here is an adapted fragment of the script I use (my actual script serves multiple cameras and does some additional sanity/error checking, archive cleaning, and logging)

```
# Note; this is a bash script for FreeBSD, toolpaths will differ on linux; see comments
# 'convert' is part of the ImageMagick toolset

set URI= http://1.2.3.4/capture/
set CAMERA = Workshop                     # camera name! No spaces...
set TIME = `date "+%d-%b-%Y"`
set DATE = `date "+%H:%M"`
set CONVERT = /usr/local/bin/convert     # /usr/bin/convert on linux
set WGET = /usr/local/bin/wget                # /usr/bin/wget on linux
set CAMHOME = /home/cam/$CAMERA/

$WGET --no-check-certificate --no-verbose "$URI" --output-document=$CAMHOME/capture.jpg

if ( $? == 0 ) then
    convert -quiet $CAMHOME/capture.jpg -gravity South -pointsize 16 -fill White -undercolor Black 
-annotate +0+10 "$CAMERA    $DATE $TIME" $CAMHOME/webcam.jpg
else 
    cp -f $CAMHOME/webcam-fail-image.jpg $CAMHOME/webcam.jpg
endif

cp $CAMHOME/webcam.jpg $CAMHOME/archive/$DATE-$TIME.jpg
```

easytarget, 2021-01-23:
See also #38

Semmu, 2021-04-11:
also take a look at [`fswebcam`](https://manpages.ubuntu.com/manpages/bionic/man1/fswebcam.1.html), as it can do timestamping too.

flaviopuhl, 2021-11-22:
Hello, any idea how to include the timestamp on the picture taken by ESP32 CAM?

easytarget, 2024-12-09:
Still a good idea, but wont be worked on now.
I am leaving this issue open for reference; although I am archiving the project

LucidEye, 2025-11-04:
Has the timestamp feature been added to this firmware?  I also need an ESP32-CAM that can natively include a date and time stamp on the jpeg image stream.  I'm really surprised that there are no ESP32-CAM firmware projects that include this feature... a simple timestamp is a basic, standard, and necessary feature of pretty much all security camera software.