easytarget/esp32-cam-webserver #88: project abandoned?

TungstenE2, 2021-02-24:
@easytarget are you still working on this project? Please do not let it go down....

easytarget, 2021-03-04:
Hi, It's Not Dead, just Resting.
I've got it to a state where it works for my needs. And I needed to do some other [stuff](https://github.com/easytarget/LaserWeb4) while dealing with 'the current situation'. 

Expressif releasing v1.0.5 of their Arduino toolchain has prompted me to look briefly at this; I've made sure it compiles and runs properly (no problems seen, I just spent a couple of hrs idly doing that and upgrading my home cams). It contains a lot of small fixes, and is something I'd been waiting to happen. The PlatformIO build has been using a lot of these fixes for some time since their library is updated more consistently. But I want to keep this project compatible with the official IDE and so I had to wait and hope.

I'll get the final 3.0 release out later today, and try to return to this project if my time allows. If anybody wants to submit PR's I'll respond quickly.

easytarget, 2021-03-07:
Note: I have 3.0 out now, I had a kick the Travis build kicking and screaming into the Python3 light first.

closed 2021-03-07 by easytarget