easytarget/esp32-cam-webserver #89: Quality settings are wrong

TungstenE2, 2021-02-24:
Quality settings are wrong.

For all settings the lower settings are on the left and higher settings are on the right.

For Quality the better quality is on the left and lower quality is on the right.
This should be changed?

Also what is the meaning of 10 and 63?

[image]

easytarget, 2021-03-04:
I have little idea what the Quality settings actually mean, I suspect the answer is buried in the Expressif documentation or some Youtube channel.
But you are right about the inconsistent slider. Humm.
https://stackoverflow.com/questions/27794439/html5-input-type-range-from-positive-to-negative
aargh. I'll see if I can slip this in.

easytarget, 2021-03-07:
fixed by: 13f193d6f063857e90e6bddc902ccc22c309a60d

TungstenE2, 2021-05-09:
ther is still a small issue to fix:

Mouse over still say: Lower is better, which is now no longer valid.

May be even 10 and 63 shoudl also be adjusted, as the numers do not make sense...

[image]

easytarget, 2021-05-10:
Yes. It says that because it is correct. The setting is 'Quality' and lower numbers are better quality. I think it's clear enough but might add a note that 'better' requires more network bandwidth.

The range of values is dictated by ESP [see here](https://github.com/espressif/esp32-camera/blob/master/driver/include/sensor.h#L116) and practicality. Values above 63 are not allowed, values below 10 provide no additional benefit. 

You can see this yourself by sending other values using the API and looking at the framesizes in the serial debug log; what I saw at a resolution of 800x600 was that at quality 63 the frames are only 6k in size, at quality 30 they are 12k,  at quality 10 it is 22k, quality 1 was a whopping 210k per frame and quality 0 things stopped working. Values above 63 were ignored.

I'm not going to change the current range, if you want values below 10 please use the API.

TungstenE2, 2021-05-10:
Ok, may be I was not clear enough.

Currently 63 on the left is giving lower quality but is faster. 
10 on the right side is giving higher quality but is slower.

Still not sure what 63 (lower quality) and 10 (higher quality) exactly mean in this context.

As the line titel is quality, I would suggest just to replace the values with +/-.
Replace 63 with +
Replace 10 with -

This would be much mure user friendly,

The mouse over I would rename to 'Lower is faster'.

easytarget, 2021-05-10:
I'd already made a change to the mouseover text (since I'm going to have to do a quick 3.1.1 anyway (#106), sigh) 
But emphasizing 'Faster and Slower' is a good idea, I was waffling along about 'WiFi bandwidth'. Thanks for the clarification :-), watch this space..

easytarget, 2021-05-11:
Resolved for now by #109

closed 2021-05-11 by easytarget