easytarget/esp32-cam-webserver #90: fps and kbps settings

TungstenE2, 2021-02-24:
Would it be possible to get settings like:

Bit rate;  kbps
max frame rate: fps
kea frame interval
Bit rate control: CBR/VBR

sample from other webcam:
[image]

easytarget, 2021-03-04:
Possibly. It's a good idea, but I'm not sure how they implement it, and if their implementation works with the current way this project does streams. 
However, if it's still a good suggestion, and if moving to a different streaming system (eg a multiclient stream server) then it may be much easier to implement.
PS: Is this other cam also ESP32 based? if so, do you have a git link to the source?

TungstenE2, 2021-03-04:
no, the screenshot is from a normal webcam I use, not ESP32 related

easytarget, 2021-03-07:
I suspected as much, bandwidth calculation and limiting etc is non trivial and firmly the domain of codec writers, chip designers etc.. Unless someone else writes a stream server for the ESP that supports this it will not be possible to do these sorts of settings.

Lowering the Quality slider and Resolution will have the same effect of limiting the bandwidth; but I cannot quantify this.

closed 2021-03-07 by easytarget