easytarget/esp32-cam-webserver #92: Debug in ap mode.

Captainoddvious, 2021-03-08:
Debug toggle not functioning in ap mode.

easytarget, 2021-03-08:
Worth noting; the 'debug' option does one thing; it dumps some data about the currently running stream. If you are ttrying to debug WiFi issues this option is not relevant, the WiFi code dumps a lot of 'debug' info by default, there is nothing extra to be shown.

However, you are absolutely right. You cant toggle debug from the keyboard in AP mode; this is a bug.
As a workaround you can enable debug in your custom config; look at/near line 148 in your copy of [`myconfig.h`](https://github.com/easytarget/esp32-cam-webserver/blob/e253cde463e2f4b99041e6c1854530f1a4f9c711/myconfig.sample.h#L148)

easytarget, 2021-05-09:
I have fixed the AP mode debug toggle
The latest release (v3.1) also add the option to generate a status dump at any time in the serial console, send a 'd' to see it. It returns the same info as the web based 'Dump' page.

closed 2021-05-09 by easytarget