easytarget/esp32-cam-webserver #94: v3 - Camera module failed to initialise

TungstenE2, 2021-03-09:
Hi @easytarget,

last night I update 1 of my 3 ESP-32 cams to v3.

Everything was working fine.

This morning when I tried to connect to the stream again this failed. Webserver was sill working and LED can still be controlled.
But no stream or still image working. The cam is powered 24/7.

First I thought this was related to https://github.com/easytarget/esp32-cam-webserver/issues/85

I rebooted several time also power off/on.

Then I got this error message once:
[image]

Did another reboot and websever is up again, but still no image or stream.
[image]

Any advice? I only updated with same settings in config last night and old version was working for several months stable.

thx

TungstenE2, 2021-03-09:
very strange, I did several reboots this morning, but only the websver was working without streaming or still image.

Just opened the housing, took out the cam, attached to a power bank and cam was streaming again.

Put it back to the housing, attached it to the wall charger and it was still working.

Will need to observe the upcoming days.

easytarget, 2021-03-09:
Very much my experience with this error... I have had cam modules get 'stuck' giving it, only to work perfectly a few hrs later. Unplugginhg from the power and leaving for a while /seems/ to play a part, but that may be my imagination.

However, Power supply definitely plays a part, I've got a USB power/voltage monitor on my programmer. On my elderly desktop I see <5v on the USB, speckles on my video, wireless dropouts and occasional initialise errors. On my netbook I have 5.1V on the USB at rest, and a much better experience with video, errors, and WiFi.

It is worth noting that I run my workshop cam from a 6V wall-wart supply from a cordless screwdriver. Lots of power and voltage, the AI-Thinker boards have a power regulator onboard, so this is safe.

pipe89, 2021-03-14:
I had the same issue with my AI thinker. I solved it by commenting:
//  #define LED_PIN           33 // Status led
//  #define LED_ON           LOW // - Pin is inverted.
//  #define LED_OFF         HIGH //

in camera_pins.h under #elif defined(CAMERA_MODEL_AI_THINKER)

However, the web server has issues with Safari, but works well on Chrome.

TungstenE2, 2021-04-21:
I have again the problem with 1 of my 3 cams, that webserver is working but stream or still image is not working. 

@pipe89 why does commenting off LED options affect this?

TungstenE2, 2021-04-23:
tried commenting as suggested but still the sill image or stream are not working, while webserver is up and working.

any other ideas?

[image]

easytarget, 2021-05-08:
@pipe89  I can see no real way that disabling the status lamp would affect anything, it'as a 20mA load and that wont cause voltage problems. Maybe your board is differently wired or otherwise problematical.
Disabling the main Lamp can be useful, since it draws so much power it can undervolt a poorly supplied board.

@TungstenE2 If just one board is affected then it is quite possible that one of your camera modules has failed. Or just that board has some other issue. Software or config. Or just that something is holding the connection permanently open n it preventing anything else from connecting. It's impossible for me to debug when the symptoms are consistent with multiple possible issues.

closed 2021-05-09 by easytarget