easytarget/esp32-cam-webserver #95: Documentation

TungstenE2, 2021-03-09:
Dear @easytarget,

could you may be update your documentation on best practise in order to use Aduino IDE for your project?

Not sure if I do the steps in the right order.

- I download the code as zip file from github
- export from zip file to folder on my PC
- open Arduino IDE (latest stable version 1.8.13)
- drag and drop files to Arduino IDE
[image]
- edit myconfig.h
- compile and upload

But  by doing as above, I get a subfolder named 'esp32-cam-webserver' containing all the files duplicated.

[image]

So there should be a better way. Can you please 

[image]

https://github.com/easytarget/esp32-cam-webserver

[image]

easytarget, 2021-04-30:
Sorry slow response, but thankyou for reporting this, it's good to have someone else reading stuff and feeding back. Even if I cant always fix things properly. 

I'll add some notes to the readme and the hackaday page.

In the past I dealt with this by repacking the tarballs, but I have a cunning plan for refactoring the V4 branch in a better way (one that I see other platformIO projects using).

easytarget, 2021-05-08:
Fixed by commit 15018a5

commit 15018a5d (2021-05-08): Document renaming archive after unpacking

fix #95
closed 2021-05-08 by easytarget