easytarget/esp32-cam-webserver #98: Wrong resolution in metadata

bernardodesousa, 2021-03-26:
Great initiative, @easytarget!

Regardless of the resolution set on the controls, ffmpeg always sees the resolution 480x320 in the metadata of the stream.

[image]

Is anyone else observing this?

rolfie23, 2021-04-24:
Yes, the showed resolution is always 2 steps lower.
UXGA (1600*1200) outputs a stream of 1024*768
SXGA (1280*1024) outputs a stream of 800*608

and so on....
Is there any explanation and solution for it?

rolfie23, 2021-04-24:
It has something to do with the DCW button. I had it off, switching resolutions shows above problem. Switching DCW button on and of you have the righ resolution.

giorge1, 2021-04-25:
I don't know if this might be related to the same problem i had, but it seems that espressif added other resolutions, changing the enumeration definition of the resolutions. The GUI in this repo is using the old framesize enum, hence, when you select SVGA, it will select a different (lower) resolution. This may be the casue of your problem.
 To solve this, you could:
 * Use an older version of the esp32 libraries (i think 1.0.4 in the Arduino IDE should work fine)
 * Change the GUI code with the new enumerations value. [This is the current file, see `framesize_t` enum](https://github.com/espressif/arduino-esp32/blob/22b427df0fc61b44a48769378a105745c4683a81/tools/sdk/include/esp32-camera/sensor.h), while [this is the older version that the gui is referring to](https://github.com/espressif/arduino-esp32/commit/22b427df0fc61b44a48769378a105745c4683a81#diff-aa521a4ece89c68dd094eb6682872bdba10f4c69a058292f8c7ca60225816373). I replaced with the following in the `index_ov2640.h` file, both in simple mode and in full mode

```
  <option value="13">UXGA(1600x1200)</option>
  <option value="12">SXGA(1280x1024)</option>
  <option value="10">XGA(1024x768)</option>
  <option value="9">SVGA(800x600)</option>
  <option value="8">VGA(640x480)</option>
  <option value="6">CIF(400x296)</option>
  <option value="5">QVGA(320x240)</option>
  <option value="3">HQVGA(240x176)</option>
  <option value="1">QQVGA(160x120)</option> 
```

This might not be the optimal solution, since future changes will cause the same problem.

easytarget, 2021-04-30:
@giorge1 :+1:  Thankyou, I hadn't seen that. :eyes: 

I did check everything compiles and updates with the 1.6 expressif release, and I even scrolled through the release notes(*) but this failed to catch my attention. 

I'll have a look once my desktop has recovered from the trauma of a Fedora34 upgrade (Hey mom! I broke Esptool *and* platformIO in one go!), I suspectthe fix is trivial.

(*) Ok, the changelog. I have been a release manager for a number of embedded projects, I have opinions about this..

easytarget, 2021-04-30:
> This might not be the optimal solution, since future changes will cause the same problem.

Yeah, there are a number of controls that are currently passed between the WebUI and server as numeric values, when it would be better to pass them as enums, the 'future proof' fix is to pass and store `FRAMESIZE_QVGA` etc..

.. but I have never gotten around to figuring that out, debugging, testing and then changing the relevent key/value pairs.

easytarget, 2021-05-03:
Fixed for all the viewers.

I also took the opportunity to add some new frame sizes; a 96x96 thumbnail, 480x320 HVGA, 1280x720 HD and, for 3MP+ camera modules, 1920x1080 FHD.

I also updated Docs and default config to reflect this.

I'm going to do a 3.1 release for this, and fix up install documentaion at the same time.

closed 2021-05-03 by easytarget