easytarget/esp32-cam-webserver #99: Add Telegram integration [enhancement]

schumar, 2021-04-05:
It would be nice to have a Telegram bot running on the ESP32-Cam, as this would enable the user to receive a picture even if outside of the LAN, without configuring port-forwarding (and in most cases also dynDNS).

A very basic functionality of "take a photo (with the current settings) and send it" would suffice, i.e. user sends the message `/photo` to the bot, and it replies with a photo.

See e.g. https://randomnerdtutorials.com/telegram-esp32-cam-photo-arduino/ or https://www.survivingwithandroid.com/telegram-esp32-cam-send-image-arduino-ide/

easytarget, 2021-04-30:
I agree; I probably don't have time (or inclination) to attack this myself, but if anyone wants to give it a try I'm open to PR's.

asmaklad, 2021-06-15:
This project covers this requirement  and more pretty much : https://github.com/asmaklad/TeleView
please check it out.

easytarget, 2021-06-17:
Thank you @asmaklad ; that certainly looks like a more telegram-integrated project. I'm going to close this issue now since I want to keep th eproject focussed on being a nice example of a browser based webcam. This request is rather out-of-scope.

closed 2021-06-17 by easytarget