easytarget/esp32-cam-webserver #100: Does not work in Synology

michaelscl, 2021-04-25:
Camera stream does not work in Synology. I have the URL set correctly. Everything works in VLC.

Where could the problem be?

[image]

best regards

Michels

easytarget, 2021-04-30:
That's a pity, but I have No idea why synology fails to view this. 

It is probably just more fussy than VLC. Lots of stuff like metadata is broken for the stream part of this code. 

I'm sorry but I cant really help (although the stream code may improve in the future), The goal of this project is to make a web based player, supporting external viewers is beyond the scope I can support. 

If you really need a solution I think this is probably a question better asked of the Synology developers.

closed 2021-04-30 by easytarget

michaelscl, 2021-05-05:
Yes I understand. I was just wondering if anyone had a solution.

easytarget, 2021-05-10:
The stream provided by this firmware is a MJPEG stream, so you should set any options related to that, but afik you are doing it right 

I just looked at a couple of 'how to set Synology for Mjpeg Streams' pages I found on google; and your config should be right. Synology should be able to retrieve the stream, so don't give up hope ;-)

I'd probably adding the trailing `/` to the URI you enter (if you have not already tried that) just in case it is significant.

michaelscl, 2021-05-21:
The slash doesn't help.

It works in VLC.
Stream info:

[image]

Is the information about the MJPG codec correct? Shouldn't it be MJPEG?