easytarget/esp32-cam-webserver #107: Face detection broken (heap crash and panic) with latest ESP IDE

easytarget, 2021-05-10:
Testing for #106 turned over an antheap.
It looks like Face detection is actually broken with the latest ESP core/IDE release giving heap errors eg:
```
CORRUPT HEAP: Bad head at 0x3ffe8450. Expected 0xabba1234 got 0x00000008
abort() was called at PC 0x4008a6cd on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008e2c8:0x3ffe7150 0x4008e541:0x3ffe7170 0x4008a6cd:0x3ffe7190 0x4008a7f9:0x3ffe71c0 0x4010215b:0x3ffe71e0 0x400fdc35:0x3ffe74a0 0x400fdb25:0x3ffe74f0 0x400939a5:0x3ffe7520 0x400829f6:0x3ffe7540 0x4008a5c5:0x3ffe7560 0x4000bec7:0x3ffe7580 0x400d29ae:0x3ffe75a0 0x401292a5:0x3ffe7750 0x401299ad:0x3ffe7780 0x40129a65:0x3ffe7810 0x40129e38:0x3ffe7830 0x40128524:0x3ffe7850 0x4012857b:0x3ffe7890 0x400901d2:0x3ffe78b0

Rebooting...
```
And so forth... I also saw a watchdog error that was transient but looked related too.

A bit of research later:
* https://github.com/espressif/arduino-esp32/issues/5128
* https://esp32.com/viewtopic.php?t=14466
* https://forum.arduino.cc/t/heap-corruption-using-esp-eye-and-free/641577

I'll be able to look at this later today or tomorrow. But more hopeful I can find a fix now

easytarget, 2021-05-10:
I'll be doing a 3.1.1 hotfix release for this tomorrow.

closed 2021-05-10 by easytarget

lamnguyenth9, 2023-06-06:
have you fix the issuse yet?

PR #108 (2021-05-10): face detect fix
fix #107
commit f97d56e8 (2021-05-10): face detect fix (#108)

Fixes #107; heap corruption and crash happening with ESP 1.0.6 when face detection is enabled.