easytarget/esp32-cam-webserver #125: Both 3x and 4 fail to compile jsonlib.cpp in Arduino 1.8.1.3 and 2.0.0-beta.5

hylioja, 2021-06-09:
Both versions (3.x and 4) fail with this error message in both Arduino 1.8.1.3 and 2.0.0-beta.5:

sketch\src\jsonlib\jsonlib.cpp: In function 'String jsonExtract(String, String)':
sketch\src\jsonlib\jsonlib.cpp:75:34 error: 'std:string' has not been declared
   if (json.indexOf(name) == std::string:npos) return json.substring(0,0);

easytarget, 2021-06-14:
That looks like a really fundamental corruption in your ESP arduino core install. The `std:string` stuff is included via `Arduino.h` and a core part of all arduino toolchains.
* Upgrade to the latest Arduino IDE, currently 1.8.15
* Reinstall the ESP32 core: at v 1.0.6: https://github.com/espressif/arduino-esp32#installation-instructions
* Try the simple strings test example at: `examples -> built-in examples -> 08.Strings -> StringLength` *with the correct ESP32 board selected* in the boards list. This should compile correctly.

There is nothing to 'fix' in my code for this, you need fix whatever is wrong in your toolchain.

blurfl, 2021-06-30:
I too am seeing this issue when using the alpha preview version of the ESP arduino core. Dropping back to the 1.06 release compiles correctly. I find that the simple strings test example compiles correctly with the alpha (maybe because it does not call std::string::npos ?) At any rate the issue with std::string::npos seems to lie with the alpha version of the core.

For now, to keep working with the alpha I work around the issue by editing src/jsonlib.cpp line 76 to avoid the issue:
```
  if (json.indexOf(name) == name.length()) return json.substring(0,0);
```

easytarget, 2021-06-30:
https://www.cplusplus.com/reference/string/string/npos/
It's a constant, usually -, that represents 'No Position'

So the conditional `(json.indexOf(name) == std::string:npos) ` is basically testing to see if `name` exists in the input, I'm not convinced your modified line will work reliably.

Looks like this needs to be taken up as a issue with the ESP alpha.

easytarget, 2021-06-30:
https://github.com/espressif/arduino-esp32/issues/5342

closed 2021-06-30 by easytarget