easytarget/esp32-cam-webserver #126: Can't connect to raspberry configured as hotspot

Alexplose, 2021-06-16:
Here is the log I get :
```
Connecting to Wifi Network 1: [DC:A6:32:36:8F:7D] OpenAuto 
...........Client connection Failed
Starting WiFi
```

How can I get more information about why connection failed, other clients can connect.

Wifi key were 1234567890 and tried WXCVBNQSDFGH

I tried WXCVBNQSDFGH because the ESP can connect to another access point with this key.

So the key is not the problem.

Alexplose, 2021-06-16:
I'll try defining WIFI_WATCHDOG and adjust it to see if it connects.

easytarget, 2021-06-17:
Unfortunately you are limited to the output the ESP WiFi tool-chain gives you. If an actual error was encountered you would see it on the log. But timeouts and rejections are, afik, not reported. 

You are more likely to find usable logs on the Pi; assuming you are using HostAPD to form the hotspot there may be info in syslog, or follow [this](https://www.raspberrypi.org/forums/viewtopic.php?t=92859#p1313712) to generate a more detailed separate log. The Esp-CAM is known to be quite fussy about wifi connectivity etc, especially if a bit low on volts, so 'other devices work'; isn't quite normal. I've seen it myself when testing on many occasions, it tends to connect after a few cycles through connect/wait loop. Sometime fully de-powering the ESP helps when it's being really fussy.

For reference, my workshop cam connects to a PI running a local 'workshop' network via HostAPD and a 18m ethernet cable, it's line of sight, about 4m distance. Connectivity between the two is always instant and excellent.

Alexplose, 2021-06-21:
Finally it's ok with increased watchdog.

closed 2021-06-21 by Alexplose

easytarget, 2021-06-30:
Thanks for coming back with your solution and closing this :-)