easytarget/esp32-cam-webserver #137: dl_lib_free calls in 3.2 branch prevent compilation

tobiasisenberg, 2021-07-11:
I just tried to compile the ```3.2``` branch. In file ```app_httpd.cpp``` there are two groups of ```dl_lib_free()``` calls, which are not known on my system:
```
'dl_lib_free' was not declared in this scope
```
These calls thus prevent a successful compilation. If I comment them out the branch works, but this is likely harmful as some memory does not get freed when needed. I also tried the ```master``` branch, which works fine out of the box.

easytarget, 2021-07-11:
This indicates there is something wrong with your ESP arduino core install, or board selection.
Please doublecheck the sections on installing and selecting them in the README.
For reference: `dl_lib_free` is defined in `tools/sdk/include/esp-face/dl_lib_matrix3d.h` in the 1.0.6 ESP32 arduino core, and `tools/sdk/esp32/include/esp-face/lib/include/dl_lib_matrix3d.h` in the 2.0 preview.
The master branch no longer has face recognition, so these calls are not present.

tobiasisenberg, 2021-07-11:
Nevermind. The reason was that I was still on ```arduino-esp32``` version ```1.0.3``` due to requirements for another project (starting with ```1.0.4``` the connection times to Wifi are greatly delayed; but this is really only important for battery-operated modules). When updating to ```1.0.6``` all works as expected.

closed 2021-07-11 by tobiasisenberg

easytarget, 2021-07-13:
Thanks for updating and closing this, it's appreciated :smile: