easytarget/esp32-cam-webserver #151: add ESP32 temperature to GUI

TungstenE2, 2021-07-29:
Tasmota implemented ESP32 temperatue reading to GUI.

Is this possible to add this value to esp32-cam-webserver GUI as well?

[image]

haraldtux, 2021-08-08:
maybe like this (MCU temperature :....) :
[image]
In file : app_httpd.cpp
line 78 to add :
// the_hara 
#ifdef __cplusplus
extern "C" {
#endif
uint8_t temprature_sens_read();
#ifdef __cplusplus
}
#endif
uint8_t temprature_sens_read();
// the_hara 

line 560 to add :
//the_hara 
    int McuTc = (temprature_sens_read() - 32) / 1.8; // celsius
    int McuTf = temprature_sens_read(); // fahrenheit
    d+= sprintf(d,"MCU temperature : %i *C, %i *F<br>\n", McuTc, McuTf); 
//the_hara

easytarget, 2021-08-21:
I've already discussed and decided not to do this.
The temp sensor is not part of the cheap ESP32S1 chips most of the AI-Thinker clones are based upon. It certainly is not present in any of mine; I have tested them. See https://github.com/espressif/arduino-esp32/issues/2422
Also, dup. of #56

TungstenE2, 2021-08-23:
To be honest, this is a petty. On my cams this is working great and also Tasmota is offering this. Why not add it to the dump page? If the sensor is not present just no value is shown, but it gives more details and in my case very usefull as well.

Please reconsider. Thx

[image]

easytarget, 2021-09-02:
You are, of course, quite right. I'll see what I can do since adding this (if available) to the data page is trivial

closed 2021-09-03 by easytarget

mrneutron42, 2023-06-26:
How do we turn on the temperature reading in the GUI?
I've got Tasmota ESP32-CAM 12.5.0.3 loaded and it's only showing framerate in the GUI.

PR #156 (2021-09-03): Temperature and OTA feedback
Give the MCU temperature; fixes #151
Also show the OTA status and password on serial (but not on the dump page for security reasons)