easytarget/esp32-cam-webserver #152: stream gets blurry

TungstenE2, 2021-07-29:
I have a cam set up in a bird nesting house. During the cause of the day the stream gets blurry/washed and fine again in late afternoon.

Not sure if this is related to temperature or sun radiation. Anybody else discovered this and may be a fix?

[image]

[image]

TungstenE2, 2021-08-09:
I was able to monitor the temperatur as suggested here https://github.com/easytarget/esp32-cam-webserver/issues/151

In 3d printed housing with lamp at 90% I see even on a cloudy day with outside temperature of 24 C over 90 C on ESP32.

[image]

easytarget, 2021-08-21:
I'm afraid there is nothing I can really do here, beyond suggesting some sort of cooing, I added a Heatsink(*) to the ESP32 that runs my CNC just to be on the safe side.

But blurring in the camera is probably condensation or some side-effect from the CCD array. Exacerbated by being cose to the MCU. Maybe the camera module itself can be moved away or insulated from the ESP32 itself?

(*) Small one intended for RasPi cpu, directly bonded with thermal glue.

easytarget, 2021-09-03:
There is now, at least, a temperature feedback on  the info page.

closed 2021-09-03 by easytarget