easytarget/esp32-cam-webserver #160: HowTo: Compile esp32-cam-webserver with platformio

ddTech, 2021-09-22:
In case someone else needs it, here is what I did in order to compile the project with platformio: 

1. Download the zip file and extract it to a temporary folder
2. Create a new project in platformio and select the appropriate board. In my case the "ESP32-CAM AI Thinker"
3. replace the code of the automatically generated **main.cpp** with the content of the **esp32-cam-webserver.ino**. You might as well move the esp-32-cam-webserver.ino to the src directory, change its extension to cpp and delete the main.cpp, but I did not test that.
4. create three subfolders **jsonlib**, **parsebytes** and **storage** in **lib**
5. move **&lt;tempdir&gt;\src\parsebytes.cpp** and **parsebytes.h** to **lib\paresbytes**, the contents of **&lt;tempdir&gt;\src\jsonlib** to **lib\jsonlib** (or just move the complete folder) and **&lt;tempdir&gt;\storage.cpp** and **storage.h** to **lib\storage**
6. move the remaining files from **&lt;tempdir&gt;\src\** and **&lt;tempdir&gt;\** including **app_httpd.cpp** to **include**
7. The structure now should look like this:

[image]

8. Now remove the former subdirs from the include references in main.cpp. **#include &lt;src/parsebytes.h&gt;** has to become **#include &lt;parsebytes.h&gt;**. Search through the document, as not all references are placed upfront. **#include &lt;src/version.h&gt;** is a bit further down.
9. repeat that with **app_httpd.cpp** and **storage.cpp**
10. I added an **#include &lt;Arduino.h&gt;** to **main.cpp**. Not sure if this is necessary, but platformio generally likes it, when using the Arduino framework.
11. If I did not forget anything, this should now compile with two or three warnings. One because we did not provide a myconfig.h thus forcing the code to fallback to defaults, which is OK the other from within the framework.
 
Regards 

Frank

easytarget, 2021-09-23:
Thankyou, nicely written up, but I respectfully disagree, it is a very long-winded way of forcing the code into a very specific default PlatformIO architecture & structure. But I think that is totally unnecessary, and will make it really hard to update / merge / create PR's etc.

This project already builds on PlatformIO straight out of the box.

If you look you will see a `platformio.ini` file is already present, and as part of Travis CI testing I simply `cd` into cloned repo and execute `pio run`, which builds the code without error (unless I break it.. which is the whole point of using CI in the first place..), see `.travis.yml` for the full steps.

So, Why not; instead:

Open VSCode->platformio and select 'Open Project' from the 'open' tab. (No need to select the 'import arduino project' tool)

[image]

Browse to and select the folder where you have cloned the repo:

[image]

You now have the project in the IDE, browseable etc, and it will compile & run just fine.

[image]

closed 2021-09-23 by easytarget

easytarget, 2021-09-23:
Actually, thinking about this the partition scheme probably needs defining for the board now; the above is still using the AI-Thinker partition scheme, which will break OTA.. I'll use PlatformIO as my development IDE for a bit and make sure uploading works ;-)

But my primary focus will remain the Arduino IDE, as I've said before, I'm happy to support platformio but it's on tier 2.

ddTech, 2021-09-24:
Hi, 

> ... it is a very long-winded way of forcing the code into a very specific default PlatformIO architecture & structure. But I think that is totally unnecessary, and will make it really hard to update / merge / create PR's etc.
> 
> This project already builds on PlatformIO straight out of the box..... 

You're absolutely right, my bad.

I read that You're concentrating on the Arduino IDE - which is fine - and actually saw the platformio.ini but did not realize the "src_dir=./" entry. 
As I found no instructions for PIO, I opened the .ino file to investigate what's going on and already then PIO complained not willing to work with ino files. 

That's why I started from scratch and did not even try File -> Open Folder... 

I did that now, and yes, it compiles fine. Watching the tree closely during compile You actually see the cpp file it briefly creates.

Only downside is that You can't modify the code without pio complaining about the ino extension.

Thanks

Regards

Frank