easytarget/esp32-cam-webserver #178: Software "Off" mode?

rdragonrydr, 2021-11-02:
I've recently purchased one of these cameras, and mine has an integrated LiPo battery charger unit but no power switch (interesting oversight on the manufacturer's part...).

I unfortunately haven't had the chance to get into the ESP32's libraries yet (I'm still learning the 8266), so I'm not really up to adding this myself, but would it be possible to add a software power switch to this in a future release?

I'm looking for two features:

- Power off via a button in the web browser interface (to deep sleep mode or equivalent)
- Power off via pressing the reset button while it's on (noinit variable or RTC memory)

With of course being able to press the reset button while it's in the deep sleep/"off" mode to awaken it again.

I'd really appreciate if you could add this as a feature at some future time, or walk me through how this works so I could add it myself.

easytarget, 2021-12-15:
In principle easy, add a 'sleep' URI to the webserver and sent the whole ESP into one of it's deep sleep modes ; also you may need to put the camera module to sleep separately.

Use an interrupt on one of the spare pins on the board to wake from the sleep and resume (so long as you re-awaken the camera and do any re-init, re-start the webserver, etc.. this may be a long list).
- It may be easier to trigger a reboot on wake, unless fast wake is really important.

There is quite a lot of info about esp32 low power modes out there, it has several different sleep modes, some of which can keep the wiFi alive etc. and I know all gpio pins support interrupts. I regularly see this discussed in passing, but it's not something I have ever programmed for on this processor.

It's not a feature that I plan to add to this example though. There are other esp32 cam firmwares out there designed to work with IOT camera systems and frameworks that might already do something like this, and may be worth checking out.

easytarget, 2024-12-09:
I am leaving this issue open for reference; although I am archiving the project