easytarget/esp32-cam-webserver #179: support for external lighting

bartgrefte, 2021-11-07:
"Control of on-board lamps"

I haven't found any mention of this, but is there support for external (not on ESP32-board soldered) lighting? Let's say one or more WS2812B LED's for example.

founfabug, 2021-11-18:
GPIO4 controls the onboard "flash" LED. You can use this pin to drive an appropriate external LED circuit, and it will mirror the operation of the onboard LED.

easytarget, 2021-12-17:
As @founfabug says; the onboard illumination LED is on GPIO4, which is also brought out on one of the Module pins.
- This is the low power side of the pin; you will need a proper arduino->led mosfet drive circuit to drive LED strips, panels etc.

The onboard LED (which has it's own onboard mosfet) will still light; you can either 'desolder' it from the board or:
- override the value from `camera_pins.h` by adding the line:
  ```#define LAMP_PIN           <NEW_PIN_NUMBER> // LED FloodLamp.```
  immediately after including the pins config, currently [line#53](https://github.com/easytarget/esp32-cam-webserver/blob/master/esp32-cam-webserver.ino#L54) in `esp32-cam-webserver.ino`
- This will let you use a different pin, since we do not use the SD card all of the pins for that are available. Make sure you do not conflict with an in-use pin for the camera, psram or programmer

easytarget, 2021-12-17:
I'm moving this to discuussions since it is a general topic that may interest others.