easytarget/esp32-cam-webserver #181: No netwerk connection with compiled esp32-cam-webserver

Patsjemoe, 2021-11-12:
Hi,
Is this program working ...as provided  .. ?  I already could not compile the program as it is,  
because  there was an error with  dl_lib_free (solved in the mean time, but had to change the content), so I wonder if it is 
really functional ?

I cannot get any network connection, despite in below , you can see that I have sucessfully  initiated the 
camera, and started the web server : 
***************************************************
esp32-cam-webserver: ESP32 camera server
Code Built: Nov 12 2021 @ 17:59:45
Base Release: 3.4
Camera init succeeded
OV2640 camera module detected
Internal filesystem contents
Listing SPIFFS directory: /
Preference file /esp32cam-preferences.json not found; using system defaults.
Lamp: 0%, pwm = 0
Starting WiFi
Known external SSIDs: None
MAC address: A4:CF:12:76:18:F8
AccessPoint mode selected in config
Setting up AccessPoint
  SSID     : mySSID
  Password : MyPSW
IP address: 192.168.0.114
Starting Captive Portal
Starting web server on port: '80'
Starting stream server on port: '81'

Camera Ready!
Use 'http://192.168.0.114/' to connect
Stream viewer available at 'http://192.168.0.114:81/view'
Raw stream URL is 'http://192.168.0.114:81/'
***************************************************************
I do not see the IP address in my router, whereas when I run a similar/previous version of this program, it is visible,
and got video stream.
Trying to run it with 192.168.4.1 also did not work. 
Any idea of why this one fails ?
thank you in advance for your help.
regards
Ludo

founfabug, 2021-11-12:
Note the line 
`AccessPoint mode selected in config`
It looks like you have the unit configured as an access point so it will not try to connect to your router. I suggest you check your configuration, mine works, connecting to my wifi without problem.

I have compiled both the latest code and the beta, without the library problems you experienced, check your software installation of the ESP board in your setup, that gave me some library errors at first. Consider a clean install of Arduino software if your environment is messed up,  the code should compile without error on a good install (tested).

Patsjemoe, 2021-11-12:
Hi founfabug,
thank you for your quick reply.
I re-commented #define wifi_ap_enable, which I thought I had to uncomment to take my ssid and psw....text is confusing, as it says : once enabled the AP ssid and password will be taken from the 1st entry in teh stationlist[] above....which is my SSID and PSW....anyway, put some // before the line and have video.
still : I only have the streaming, not the menu on the left side of the screen....I can only call up my LED menu.
        I seem to take a DHCP address instead of the one I defined. (where exactly should I put my own IP address to have a fixed 
        static address ?)
regards
Ludo

founfabug, 2021-11-12:
I have not tested static IP, however, setting the DHCP flag to false in 
`struct station stationList` (this may not be clear, but seems right from proximity of comments in code)
And the 
`ST_` variables in the following seems to be the way.

```
Static network settings for client mode
 * 
 * Note: The same settings will be applied to all client connections where the dhcp setting is 'false'
 * You must define all three: IP, Gateway and NetMask
 */
// warning - IP addresses must be seperated with commas (,) and not decimals (.)
// #define ST_IP      192,168,0,123
// #define ST_GATEWAY 192,168,0,2 
// #define ST_NETMASK 255,255,255,0
// One or two DNS servers can be supplied, only the NTP code currently uses them
// #define ST_DNS1 192,168,0,2
// #define ST_DNS2 8,8,8,8
```
I encourage you to persevere, I found this code to be the best general purpose web server of those I tested

Patsjemoe, 2021-11-17:
Hi,
Problem solved, static IP works too.
Thanks for your help
Regards
Ludo

closed 2021-11-17 by Patsjemoe