easytarget/esp32-cam-webserver #182: At least some OV2640 cameras no longer recognized with version 4.0 and ESP Core 2.0.x

tobiasisenberg, 2021-11-14:
The report by @founfabug on an [issue with their OV2640 camera not being recognized](https://github.com/easytarget/esp32-cam-webserver/issues/176#issuecomment-949108975) is exactly the same I can report: I also have a board (ESP32 CAM)/camera (OV2640) combo which runs fine with the 3.x version (at least some older version of the 3.x branch from about a year ago, have not tested the most recent updates), ESP Core v. 1.0.3 or 1.0.4, and module "AI-THINKER ESP32-CAM", but shows the same errors as reported by @founfabug for 4.0 beta 2 (most recent GitHub version), ESP Core 2.0.0 and 2.0.1, and module "ESP32 Dev Module":
```
esp32-cam-webserver: BirdCam2 (ESP32)
Code Built: Nov 14 2021 @ 16:35:56
Base Release: 4.0.beta2
E (142) cam_hal: cam_dma_config(280): frame buffer malloc failed
E (142) cam_hal: cam_config(364): cam_dma_config failed
E (142) camera: Camera config failed with error 0xffffffff

CRITICAL FAILURE: Camera sensor failed to initialise.
```

I reverted to 3.x for now to make it work and re-deployed the camera module (since I need it). Interestingly, I remembered that when I had flashed another board in the same configuration a while back (with 4.0 beta 2 but before the update to ESP Core 2.0.x) it had worked. So what I tested now is the following: Using a spare module and camera and the [4.0 beta 2 release version (code download, not git clone, because this works also with ESP Core prior to 2.0.0)](https://github.com/easytarget/esp32-cam-webserver/releases/tag/v4.0.beta2), I first tested with ESP Core 2.0.1, module "ESP32 Dev Module", and ```#define CAMERA_MODEL_AI_THINKER```, which led to this result (the camera is not recognized):
```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1420
ho 0 tail 12 room 4
load:0x40078000,len:13540
load:0x40080400,len:3604
entry 0x400805f0

====
esp32-cam-webserver: TestCam2 (ESP32-CAM)
Code Built: Nov 14 2021 @ 18:24:25
Base Release: 4.0.beta2
E (61) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)

CRITICAL FAILURE: Camera sensor failed to initialise.

A full (hard, power off/on) reboot will probably be needed to recover from this.
Meanwhile; this unit will reboot in 1 minute since these errors sometime clear automatically
Lamp: 0%, pwm = 0
Starting WiFi
Known external SSIDs:  'XXXXXXXXX' 'YYYYYYYYYY'
MAC address: mm:mm:mm:mm:mm:mm
Scanning local Wifi Networks
3 networks found
  1 : [xx:xx:xx:xx:xx:xx] XXXXXXXXX (-38)  -  Known!
  2 : [yy:yy:yy:yy:yy:yy] YYYYYYYYYY (-76)  -  Known!
  3 : [zz:zz:zz:zz:zz:zz] ZZZZZZZZZZ (-93)
Connecting to Wifi Network 0: [xx:xx:xx:xx:xx:xx] XXXXXXXXX 
.........Client connection succeeded
IP address: 192.168.4.3
Netmask   : 255.255.255.0
Gateway   : 192.168.4.1
Setting up OTA
Time functions disabled
Starting web server on port: '80'
Starting stream server on port: '81'

Camera unavailable due to initialisation errors.

No PSRAM found.
Please check the board config for your module.
High resolution/quality images & streams will show incomplete frames due to low memory.

This is the 4.0 alpha
 - Face detection has been removed!
```

Next try, with the exact same code but with ESP Core 1.0.4, but the rest identical (and now everything works):
```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

====
esp32-cam-webserver: TestCam2 (ESP32-CAM)
Code Built: Nov 14 2021 @ 18:34:42
Base Release: 4.0.beta2
Camera init succeeded
OV2640 camera module detected
Internal filesystem contents
Listing SPIFFS directory: /
Preference file /esp32cam-preferences.json not found; using system defaults.
Lamp: 0%, pwm = 0
Starting WiFi
Known external SSIDs:  'XXXXXXXXX' 'YYYYYYYYYY'
MAC address: mm:mm:mm:mm:mm:mm
Scanning local Wifi Networks
2 networks found
  1 : [xx:xx:xx:xx:xx:xx] XXXXXXXXX (-37)  -  Known!
  2 : [yy:yy:yy:yy:yy:yy] YYYYYYYYYY (-75)  -  Known!
Connecting to Wifi Network 0: [xx:xx:xx:xx:xx:xx] XXXXXXXXX 
.Client connection succeeded
IP address: 192.168.4.3
Netmask   : 255.255.255.0
Gateway   : 192.168.4.1
Setting up OTA
Time functions disabled
Starting web server on port: '80'
Starting stream server on port: '81'

Camera Ready!
Use 'http://192.168.4.3/' to connect
Stream viewer available at 'http://192.168.4.3:81/view'
Raw stream URL is 'http://192.168.4.3:81/'
Camera debug data is disabled (send 'd' for status dump, or any other char to enable debug)

No PSRAM found.
Please check the board config for your module.
High resolution/quality images & streams will show incomplete frames due to low memory.

This is the 4.0 alpha
 - Face detection has been removed!
```

Just to be sure, I tested this camera and module combo again with ESP Core 2.0.1 and the most recent version pulled from Github, and indeed the error messages match those I had reported in the beginning for the other camera/module combo (that I have deployed so I cannot test as easily):
```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1420
ho 0 tail 12 room 4
load:0x40078000,len:13540
load:0x40080400,len:3604
entry 0x400805f0

====
esp32-cam-webserver: TestCam2 (ESP32-CAM)
Code Built: Nov 14 2021 @ 19:10:00
Base Release: 4.0.beta2
E (143) cam_hal: cam_dma_config(280): frame buffer malloc failed
E (143) cam_hal: cam_config(364): cam_dma_config failed
E (143) camera: Camera config failed with error 0xffffffff

CRITICAL FAILURE: Camera sensor failed to initialise.

A full (hard, power off/on) reboot will probably be needed to recover from this.
Meanwhile; this unit will reboot in 1 minute since these errors sometime clear automatically
Lamp: 0%, pwm = 0
Starting WiFi
Known external SSIDs:  'XXXXXXXXX' 'YYYYYYYYYY'
MAC address: mm:mm:mm:mm:mm:mm
Scanning local Wifi Networks
3 networks found
  1 : [xx:xx:xx:xx:xx:xx] XXXXXXXXX (-38)  -  Known!
  2 : [yy:yy:yy:yy:yy:yy] YYYYYYYYYY (-76)  -  Known!
  3 : [zz:zz:zz:zz:zz:zz] ZZZZZZZZZZ (-93)
Connecting to Wifi Network 0: [xx:xx:xx:xx:xx:xx] XXXXXXXXX 
..........Client connection succeeded
IP address: 192.168.4.3
Netmask   : 255.255.255.0
Gateway   : 192.168.4.1
Setting httpURL
Setting up OTA

No OTA password has been set! (insecure)

Time functions disabled
Starting web server on port: '80'
Starting stream server on port: '81'

Camera unavailable due to initialisation errors.

No PSRAM found.
Please check the board config for your module.
High resolution/quality images & streams will show incomplete frames due to low memory.

This is the 4.0 alpha
 - Face detection has been removed!
```
The camera is again not recognized and the error messages match the first ones I had reported, but the ones from the pure 4.0 beta 2 under ESP Core 2.0.1 are slightly different. But the results is the same, with ESP Core 2.0.x the OV2640 camera is not recognized.

This indicates to me that this is a general issue that applies to several modules/cameras, and that has something to do with the transition from ESP Core 1.x to ESP Core 2.x. It would be great if this could be figured out.

easytarget, 2021-12-17:
`No PSRAM found.`

Check the 4.x install instructions and ensure you have enabled the correct board and enabled psram  (it continually defaults to off, very inconvenient) then re-program.

If you still see 'no psram' messages than you have a defective/unsupported board.
It appears that the later ESP ide cores need psram to use the camera features.

tobiasisenberg, 2021-12-31:
I am confirming that this was a configuration mistake on my side. I had forgotten about the ```Minimal SPIFFS (1.9MB APP with OTA/190KB SPIFFS)``` and the ```PSRAM enabled``` settings. The newest version (```v4.0.beta3```) works like a charm with ```ESP Core 2.0.2``` now. Sorry for the confusion, and thanks for the pointer! Amazing work!

closed 2021-12-31 by tobiasisenberg

easytarget, 2021-12-31:
Thankyou for coming back and sharing that, really useful to know and really helpful :-)