easytarget/esp32-cam-webserver #183: ESP-EYE CameraWebServer Sometimes Works...Other times fails

Rakan-AlZagha, 2021-11-14:
Hey all,

I am pretty new to working with the ESP line of Microcontrollers. Recently, I was able to get the Camera WebServer from the examples on Arduino's IDE to work. However, it is very unpredictable. Every time I run the program I am able to get an IP address and it connects to my WiFi, however, I get an "ERR_CONNECTION_TIMED_OUT" error from the browser often.

The times that it does work, it will work without interruption and the video is great. Even if I reset it, it will still work without issues. However, last night I disconnected it from my computer and put it away and tried to get it back up again and NOTHING. I keep getting the timed out error even though I am getting an IP. I tried reflashing, restarting my computer, double checking wifi is up, and still nothing. To note, I am on a shared college WiFi that is fairly strong, very few breakdowns. What could be the issue here? Am I missing something? I am attaching the code and SS of my configuration. I am using the ESP-EYE DevKit for AIoT Developers.

Would appreciate any help.

Thank you.

```
#include "esp_camera.h"
#include <WiFi.h>

//
// WARNING!!! PSRAM IC required for UXGA resolution and high JPEG quality
//            Ensure ESP32 Wrover Module or other board with PSRAM is selected
//            Partial images will be transmitted if image exceeds buffer size
// FINAL

// Select camera model
//#define CAMERA_MODEL_WROVER_KIT // Has PSRAM
#define CAMERA_MODEL_ESP_EYE // Has PSRAM
//#define CAMERA_MODEL_M5STACK_PSRAM // Has PSRAM
//#define CAMERA_MODEL_M5STACK_V2_PSRAM // M5Camera version B Has PSRAM
//#define CAMERA_MODEL_M5STACK_WIDE // Has PSRAM
//#define CAMERA_MODEL_M5STACK_ESP32CAM // No PSRAM
//#define CAMERA_MODEL_M5STACK_UNITCAM // No PSRAM
//#define CAMERA_MODEL_AI_THINKER // Has PSRAM
//#define CAMERA_MODEL_TTGO_T_JOURNAL // No PSRAM

#include "camera_pins.h"

const char* ssid = "*******";
const char* password = "*******";

void startCameraServer();

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.println();

  camera_config_t config;
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sscb_sda = SIOD_GPIO_NUM;
  config.pin_sscb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.pixel_format = PIXFORMAT_JPEG;
  
  // if PSRAM IC present, init with UXGA resolution and higher JPEG quality
  //                      for larger pre-allocated frame buffer.
  if(psramFound()){
    config.frame_size = FRAMESIZE_UXGA;
    config.jpeg_quality = 10;
    config.fb_count = 2;
  } else {
    config.frame_size = FRAMESIZE_SVGA;
    config.jpeg_quality = 12;
    config.fb_count = 1;
  }

#if defined(CAMERA_MODEL_ESP_EYE)
  pinMode(13, INPUT_PULLUP);
  pinMode(14, INPUT_PULLUP);
#endif

  // camera init
  esp_err_t err = esp_camera_init(&config);
  if (err != ESP_OK) {
    Serial.printf("Camera init failed with error 0x%x", err);
    return;
  }

  sensor_t * s = esp_camera_sensor_get();
  // initial sensors are flipped vertically and colors are a bit saturated
  if (s->id.PID == OV3660_PID) {
    s->set_vflip(s, 1); // flip it back
    s->set_brightness(s, 1); // up the brightness just a bit
    s->set_saturation(s, -2); // lower the saturation
  }
  // drop down frame size for higher initial frame rate
  s->set_framesize(s, FRAMESIZE_QVGA);

#if defined(CAMERA_MODEL_M5STACK_WIDE) || defined(CAMERA_MODEL_M5STACK_ESP32CAM)
  s->set_vflip(s, 1);
  s->set_hmirror(s, 1);
#endif

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");

  startCameraServer();

  Serial.print("Camera Ready! Use 'http://");
  Serial.print(WiFi.localIP());
  Serial.println("' to connect");
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(10000);
}
```
[image]
[image]
[image]

founfabug, 2021-11-15:
Hello, this looks like an http: / https: issue (I had this). Some browsers automatically change the http to https, and do this silently. ESP32 has no service for the https port, so the "site can't be reached". Try changing back to http:// in the browser. If that fails (which it does on modern versions of firefox, for example, as it continually changes to https), then bookmark the https site (the one that can't be reached), and edit the bookmark to point to the http site.  Use the bookmark and all is well.

Rakan-AlZagha, 2021-11-15:
Hello @founfabug , I just tried it and unfortunately it does not seem to be working still. I saved it as a bookmark and still nothing :/, any other suggestions?

founfabug, 2021-11-15:
If the log on the ESP32 shows all is working, I would suspect the browser and check that first. If that does not give success I would set the ESP32 up as an Access Point (AP) and connect to it directly to ensure there is no firewall issue.

Rakan-AlZagha, 2021-11-15:
I set it up as an AP and it works flawlessly right now, it might be a firewall or timeout issue with my college's WiFi to be honest...I am going to keep the issue open if anyone has any input about the WiFi router problem...but thank you SOOOO much.

Patsjemoe, 2021-11-22:
Hi,
I faced a similar problem due to the router automatically changing between 2.4 and 5 ghz, and esp32cam needs 2.4 ghz.

Rakan-AlZagha, 2021-12-07:
So, @Patsjemoe that was exactly the issue. To everyone pulling their hair out over this issue, here is what worked for me:
- I am on a university WiFi and unfortunately I do not have access to the WiFi configuration settings, however, the issue here is that the WiFi automatically changes between 2.4GhZ and 5 GhZ. The camera web server is super spotty and would sometimes work and other times would not start no matter what I did. 
- To remedy this, I found 2 solutions:
1. well, if you have a home router, just toggle the 2.4GhZ signal when you are running your ESP device
2. if you are on a school network, or anywhere in the world frankly but do not have access to the router settings, you can use your mobile hotspot on your laptop. You have the option of creating a hotspot on your laptop and only emitting a 2.4GhZ signal. (this is what I am doing)

Hope this helps anyone else!

closed 2021-12-07 by Rakan-AlZagha