easytarget/esp32-cam-webserver #185: Webserver menu issue

Patsjemoe, 2021-11-17:
Hi,
The application works, but get only a small part of the webserver menu (see picture)
[image]
Any idea?
Regards
Ludo

founfabug, 2021-11-18:
From memory, the "Full" button opens the large menu with all options. If it does not, try refreshing, or a different browser.

Patsjemoe, 2021-11-18:
Hi,
When I push that button, I get an error... Header fields are too long....
Regards
Ludo

founfabug, 2021-11-18:
This search suggests that the user agent field is too long (it comes from the web browser), and how to work around it, by using desktop site, or a different browser:

https://duckduckgo.com/?q=esp32-cam+header+fields+are+too+long&t=fpas&ia=web

Patsjemoe, 2021-11-19:
Thank you, it works with another browser.
Still facing a few issues with android, where led does not always react to a switch off, and when doing a menu save, it is not saving.

founfabug, 2021-11-19:
Check your SPIFFS is formatting, an error in your log will tell you if it does not. If you have this problem, read my "fix" in comments in issue 175:

> ERROR: E (1099) SPIFFS: mount failed, -10025 #175

Patsjemoe, 2021-11-20:
Hi,
This problem occurs with android browser..... Not sure how to log this, as arduino is not open then.
Regards
Ludo

Patsjemoe, 2021-11-20:
I switched to Firefox, here I do not have this issue. Now trying to use https secure with dns.

closed 2021-11-21 by Patsjemoe

easytarget, 2021-12-17:
For 'header fields too long' errors see issue #1 
- Thsi is a hard limitation imposed by the expressif web server code
- Android browsers seem to add additional headers as you navigate through the UI that overflow the buffer. Basically sometimes it will work, sometimes it will not. 
  - Opening the page you want directly in a new browser tab usually gets past this
- Unfortunately expressif do not provide any way to fix this in the arduino IDE core.