easytarget/esp32-cam-webserver #186: ESP32cam DNS error due to http not secure

Patsjemoe, 2021-11-21:
Hi,
ESP32cam webserver works on the local network.
Trying to use it with DNS, I get the menu, can take a still picture, but when trying to stream, I receive an error  : 
Access to image at 'http://streamingport /' from origin 'http://xxx.ddns.yyy:menuport' has been blocked by CORS policy: The request client is not a secure context and the resource is in more-private address space `private`. 
I assume I need to add https , but cannot find any example for esp32cam...... where to find it ?
regards
Ludo

Patsjemoe, 2021-12-10:
I am surprised that all ESP32cam examples are NOT-HTTPS, specially in these days where security is important, it would be nice to have an example of the webserver with HTTPS. All applications can be hacked with http only....

easytarget, 2021-12-15:
See #54
This is an example, it is not suitable for use if you require a private stream.

closed 2021-12-15 by easytarget