easytarget/esp32-cam-webserver #187: Error while compiling

Kallemanni, 2021-11-22:
When I try to compile the sketch for uploading I get an error pointing on "app.httpd.cpp".

It says:

```
sketch\app_httpd.cpp: In function 'esp_err_t capture_handler(httpd_req_t*)':
app_httpd.cpp:411:37: error: 'dl_lib_free' was not declared in this scope
         dl_lib_free(net_boxes->score);
                                     ^
sketch\app_httpd.cpp: In function 'esp_err_t stream_handler(httpd_req_t*)':
app_httpd.cpp:527:61: error: 'dl_lib_free' was not declared in this scope
                                 dl_lib_free(net_boxes->score);
```

What do I have to do? 
Thank you and kind regards.

Kallemanni, 2021-11-23:
I mentioned that I need to update the ESP32 Core for Arduino IDE to version 1.0.6

easytarget, 2021-12-17:
You need to upgrade to esp core 2.0.x since that is a requirement in the latest versions of the cam software.

closed 2021-12-17 by easytarget