easytarget/esp32-cam-webserver #188: Cannot get lamp to illuminate with curl

stsr505089, 2021-11-25:
Loving the software, iIt works pretty well considering the shprt comings of tLoving the software, iIt works pretty well considering the shprt comings of the hardware and the various issues with different camera modules etc.    I want to take a still jpg each minute and have the lamp illuminate the scene. (inside a dark cupboard.watching electricity metes).  The only way I can get it to work is to have the lamp illuminate all the time.   It sort of works if I enable auto lamp in that it flashes when my bash script take a picture ( wget http://192.168.1.143:80/capture ) but it is usually dark or partially illuminated because the timing is off.  D=oing the obvious swtiching the lamp on ( curl http://192.168.1.143/control?var=lamp&val=100 )* pausing then taking then getting the capture fails with "This URI doesn't exist"  If I try anything else using control, it does the same.  Doing "curl http://192.168.1.143/status" works fine.   I'm using the current master vesion.    Tried all kind of things in my bash script without success
I'm maybe missing something ?

15498th, 2021-12-16:
Executing `curl http://192.168.1.143/control?var=lamp&val=100` fails because  `&` symbol gets interpreted by shell. Try `curl "http://192.168.1.143/control?var=lamp&val=100"` instead.

I tried to reproduce autolamp timing being off, but after I figured out how to make it light it worked fine. Posting serial output with debug enabled might help to figure out the issue.

easytarget, 2021-12-17:
As @15498th says; make sure you are correctly escaping the URL's for curl, 
Have a look at the return from the curl command, it will list the actual URL it tried.

I'll do a quick test myself on the lamp timings

easytarget, 2021-12-17:
Yep.. looks like the image can be captured before the lamp is fully on (there is a small 'lead' time for the pwm to settle, and another while the UVled makes the phosphor (the yellow circle) glow, I suspect the module is fast enough to beat that,. I'll experiment with a delay

easytarget, 2021-12-20:
Yep, I needed to add a little more delay to the lamp; esp themselves add 150ms, I only added 75, but I'll increase this to 150

Debugging this opened up a can of worms regarding how images sit in, and are served from, the framebuffer.

closed 2021-12-20 by easytarget